vekexasia/comfoair-esp32 #1: Connection schema for the ESP32 and CAN

asmanea, 2021-04-09:
I hope I'm asking too much! Could you please upload a connection scheme or image of the end result (not inside the box) between VMC and ESP32/CAN? The steps provided are enough for a skilled person, but for a newbie might be cumbersome to setup. Thanks!

Uriziel01, 2021-06-26:
Just run power (+12V and GND) into the step down converter (set it to output 3.3V first!) and then connect the Step down ouput 3.3V+GND into both ESP32 and the CAN Transceiver. Next connect RX/TX from CAN Transceiver into pins 4 and 5 of ESP32.
Maybe we could create a graphical form of wiring if this will help you ;)

asmanea, 2021-06-26:
Thanks! Will order the parts, and try to do the connection schema. Will update shortly.

Uriziel01, 2021-06-26:
Sure, it is pretty straightforward. I've actually done the build today, should take you around 30minutes.
My only issue was waaaay to small of a box to fit everything inside of it 😄 Now onto HomeAssistant configuration.

vekexasia, 2021-06-26:
Glad to see you did this project. The box is just enough to fit all components but you need to be good at tetris :D

Uriziel01, 2021-06-28:
BTW @vekexasia would you be willing to merge in some MQTT tweaks like implementation of HA MQTT Discovery or migrating commands into using payload instead of just topics. Currently it does not work well with HA without manual synchronization automation scripts.
Not sure if I should just implement this locally or do a PR into your repository? 😉

Uriziel01, 2021-06-28:
@asmanea I've uploaded simple connection diagram so it is easy even for a beginner. Keep in mind (as you can see in the photo) I DID NOT USED this exact diagram because I hate destroying ethernet cables without a reason so I've just plugged in cable into the RJ45 socket and the last 4 pins correspond to the pins on the left (+12V, GND, CAN_H, CAN_L), this way you use just a ethernet cable with RJ45 on both ends.

[image]

vekexasia, 2021-06-28:
@Uriziel01 ofcourse. Also  if you'd like to add the diagram! that'd be perfect. even if i used the mini format factor.

svde, 2021-06-30:
> RJ45 socket and the last 4 pins correspond to the pins on the left (+12V, GND, CAN_H, CAN_L),

How does this correspond with TIA-568B color coding?

| ComfoAir | RJ45 pin | TIA 568B |
| --- | --- | --- |
| CAN_L | 1 | white-orange |
| CAN_H | 2 | orange |
| GND | 3 | white-green |
| 12V | 4 | blue |

Niki206cc, 2021-07-05:
Good morning, i'm new with HA and ESPN32. I bought all but i want to be sure i will connect correctly because i have a different device. Can you pls confirm me the connections are ok? Thanks
[image]

svde, 2021-07-05:
> Can you pls confirm me the connections are ok? Thanks

Yes, that's correct.

pistacjusz, 2021-07-06:
> @asmanea I've uploaded simple connection diagram so it is easy even for a beginner. Keep in mind (as you can see in the photo) I DID NOT USED this exact diagram because I hate destroying ethernet [...]

Are you sure of the attached wiring diagram?
I connected the circuit as shown in the attached below photos, but it doesn't work completely. I can read the values of temperature, humidity, etc. but I cannot change the state of the recuperator (e.g. change the fan speed). I think the HA is sending the mqtt command correctly. Do you have any idea why this doesn't work? 
[image]
[image]
[image]

svde, 2021-07-06:
> but I cannot change the state of the recuperator (e.g. change the fan speed). 
Same here.

I've been trying to investigate (capturing CAN traffic with a raspberry pi + CANhat), haven't figured out what pattern to watch for in the candump output to see if the mqtt command (mosquitto_pub -h 127.0.0.1 -t comfoair/commands/boost_10_min -m 1) is making it onto the canbus.

Uriziel01, 2021-07-18:
@pistacjusz @svde Yes I'm 100% sure because it is working at my place for the last few weeks and except for a WiFi stability issues (not sure what is the problem here, just have to restart the ESP now and then) it is working just fine. Are you sure you are sending correct MQTT msg with corrent topic? Matching your defined `MQTT_PREFIX` from fw files?

pistacjusz, 2021-07-18:
@Uriziel01 On your schematic you connect Can_H recuperator to Can_L transceiver and Can_L recuperator to Can_H transceiver.
When I connect my circuit like this, nothing happens.

Below log from `mosquitto_sub -v -t 'comfoair / commands / #'`

comfoair / commands / ventilation_level 0
comfoair / commands / ventilation_level 1
comfoair / commands / ventilation_level 3
comfoair / commands / set_mode auto
comfoair / commands / set_mode manual
comfoair / commands / ventilation_level 0
comfoair / commands / boost_10_min (null)
comfoair / commands / ventilation_level_3 (null)

and second `mosquitto_sub -v -t 'comfoair / #'`

comfoair / exhaust_fan_speed 805
comfoair / supply_fan_speed 899
comfoair / away_indicator false
comfoair / exhaust_fan_flow 58
comfoair / supply_fan_flow 56
comfoair / ah_actual 0.65
comfoair / exhaust_fan_speed 810
comfoair / supply_fan_speed 901
comfoair / exhaust_fan_flow 59
comfoair / exhaust_fan_speed 813
comfoair / supply_fan_speed 902
comfoair / exhaust_fan_speed 816
comfoair / supply_fan_speed 903
comfoair / exhaust_fan_speed 818 

Here line from secrets.h file
#define MQTT_PREFIX "comfoair"

svde, 2021-07-18:
Sending the commands:

$ mosquitto_pub -t comfoair/commands/boost_10_min -m 1
$ mosquitto_pub -t comfoair/commands/boost_10_min -m 2
$ mosquitto_pub -t comfoair/commands/boost_10_min -m 2
$ mosquitto_pub -t comfoair/commands/boost_10_min -m 2

and they show up like this:

$ mosquitto_sub -v -t 'comfoair/commands/#'
comfoair/commands/boost_10_min 1
comfoair/commands/boost_10_min 2
comfoair/commands/boost_10_min 2
comfoair/commands/boost_10_min 2

I'm also running candump, shouldn't I be seeing something like this being sent over the canbus?

#define CMD_boost_10_min                    { 0x84, 0x15, 0x01, 0x06, 0x00, 0x00, 0x00, 0x00, 0x58, 0x02, 0x00, 0x00, 0x03, 0x00, 0x00, 0x00 }

Uriziel01, 2021-07-19:
@pistacjusz Haha, just noticed this :D Of course the correct method is the other way around 😄 

Both of your commands seems ok, did you tried to connect to serial of ESP32 and check if it correctly receives the MQTT message and tries to send the CAN command?

pistacjusz, 2021-07-20:
> 
> 
> @pistacjusz Haha, just noticed this 😆 Of course the correct method is the other way around, will fix the diagram before making the PR I've mentioned 😄
> 
> Both of your commands seems ok, did you tried to connect to serial of ESP32 and check if it correctly receives the MQTT message and tries to send the CAN command?

There is some data from the serial monitor after sending the MQTT message

```
supply_fan_speed - 1161-------new message from broker-----
channel:comfoair/commands/boost_60_min
data:
Received: boost_60_min
1F034051 00 84 15 01 06 00 00 00
1F034051 01 00 10 0E 00 00 03 00
1F034051 82 00 00
1F054051 00 84 15 01 06 00 00 00 
1F054051 01 00 10 0E 00 00 03 00
1F054051 82 00 00
1E4041 X 22 4
Decoded :)
exhaust_fan_speed - 1026354041 X 255 0
Decoded :)

exhaust_fan_speed - 1024354041 X 256 0 
Decoded :)
ah_actual - 0.8610000001 X 0
1000001A X 0
10000001 X 0
1000001A X 0
-------new message from broker-----
channel:comfoair/commands/ventilation_level_3
data:
Received: ventilation_level_3
1F074051 00 84 15 01 01 00 00 00
1F074051 81 00 01 00 00 00 03
1F014051 00 84 15 01 01 00 00 00 
1F014051 81 00 01 00 00 00 03
1000001A X 0
1E4041 X 28 4 
Decoded :)

supply_fan_speed - 1138-------new message from broker-----
channel:comfoair/commands/ventilation_level
data:1
Received: comfoair/commands/ventilation_level
ventilation_level_1
1F034051 00 84 15 01 01 00 00 00
1F034051 81 00 01 00 00 00 01
1F054051 00 84 15 01 01 00 00 00 
1F054051 81 00 01 00 00 00 01
688041 X 10
1E4041 X 20 4
Decoded :)

ah_actual - 0.85-------new message from broker-----
channel:comfoair/commands/ventilation_level
data:3
Received: comfoair/commands/ventilation_level
ventilation_level_3
1F074051 00 84 15 01 01 00 00 00
1F074051 81 00 01 00 00 00 03
1F014051 00 84 15 01 01 00 00 00 
1F014051 81 00 01 00 00 00 03
10000001 X 0
1E4041 X 25 4
Decoded :)

exhaust_fan_duty - 26-------new message from broker-----
channel:comfoair/commands/set_mode
data:auto
Received: comfoair/commands/set_mode
1F030051 85 15 08 01
1F050051 85 15 08 01 
1E8041 X 273 4
Decoded :)
supply_fan_speed - 11391DC041 X 253 0
Decoded :)
```

svde, 2021-07-20:
Here is my output from the serial monitor:

```
21:04:22.253 -> exhaust_fan_speed - 147710000001 X 0
21:04:22.892 -> 4C4041 X 228 0 
21:04:22.892 -> 810042 X 2C 0 
21:04:22.930 -> 4C8041 X 229 0 
21:04:23.038 -> 354041 X 2A5 0 
21:04:23.038 -> Decoded :)
21:04:23.038 -> ah_actual - 1.651DC041 X 2D0 0 
21:04:23.072 -> Decoded :)
21:04:23.072 -> exhaust_fan_flow - 2081E4041 X 2BF 5 
21:04:23.254 -> Decoded :)
21:04:23.254 -> exhaust_fan_speed - 14711E8041 X 2A2 5 
21:04:23.871 -> Decoded :)
21:04:23.871 -> supply_fan_speed - 144210000002 X 0
21:04:24.095 -> 1DC041 X 2CF 0 
21:04:24.095 -> Decoded :)
21:04:24.095 -> exhaust_fan_flow - 2071E4041 X 2C2 5 
21:04:24.278 -> Decoded :)
21:04:24.278 -> exhaust_fan_speed - 147410000001 X 0
21:04:24.888 -> 80C042 X 20 0 
21:04:24.888 -> -------new message from broker-----
21:04:24.888 -> channel:comfoair/commands/boost_60_min
21:04:24.888 -> data:0
21:04:24.888 -> Received: boost_60_min
21:04:24.888 -> 1F034051 00 84 15 01 06 00 00 00 
21:04:24.924 -> 1F034051 01 00 10 0E 00 00 03 00 
21:04:24.924 -> 1F034051 82 00 00 
21:04:25.903 -> 1F054051 00 84 15 01 06 00 00 00 
21:04:25.903 -> 1F054051 01 00 10 0E 00 00 03 00 
21:04:25.903 -> 1F054051 82 00 00 
21:04:25.903 -> 1DC041 X 2D0 0 
21:04:25.903 -> Decoded :)
21:04:25.903 -> exhaust_fan_flow - 2081E4041 X 2C7 5 
21:04:25.903 -> Decoded :)
21:04:25.903 -> exhaust_fan_speed - 147910000002 X 0
21:04:25.939 -> 810042 X 218 0 
21:04:26.156 -> 1DC041 X 2D1 0 
21:04:26.156 -> Decoded :)
21:04:26.156 -> exhaust_fan_flow - 2091E4041 X 2CB 5 
21:04:26.334 -> Decoded :)
21:04:26.334 -> exhaust_fan_speed - 148310000001 X 0
21:04:26.902 -> 80C042 X 2C 0 
21:04:26.902 -> 4C8041 X 228 0 
21:04:26.902 -> 450041 X 2D6 0 
21:04:26.902 -> Decoded :)
21:04:26.902 -> outdoor_air_temp - 21.4454041 X 2D6 0 
21:04:26.902 -> Decoded :)
21:04:26.902 -> pre_heater_temp_after - 21.4370041 X 2D6 0 
21:04:26.977 -> Decoded :)
21:04:26.977 -> pre_heater_temp_before - 21.41E0041 X 2CF 0 
21:04:27.088 -> Decoded :)
21:04:27.088 -> supply_fan_flow - 2071E8041 X 291 5 
21:04:27.088 -> Decoded :)
21:04:27.088 -> supply_fan_speed - 1425354041 X 2A1 0 
21:04:27.088 -> Decoded :)
21:04:27.088 -> ah_actual - 1.611DC041 X 2D2 0 
21:04:27.163 -> Decoded :)
21:04:27.163 -> exhaust_fan_flow - 2101E4041 X 2CD 5 
21:04:27.345 -> Decoded :)
```
but nothing happens...

Uriziel01, 2021-07-22:
I think I've just encountered similiar issue on my unit. Last night we've had some power grid brownouts and as of today I was unable to execute any commands on the unit. Logs shown that it was receiving the MQTT data but nothing was executed (but also nothing was decoded so it is not identical as your case), could you try to power cycle the unit few times?

It helped on my side, first time it did not change anything, tried for the second time - this time waiting for like 10 seconds before powering it up - and the issue resolved itself that time.

svde, 2021-08-22:
Maybe it's because my setup is a bit different? I have a Q350 (firmware R1.7.1), option box, a rpi with CAN hat and the comfoair-esp32 device.

As the command over MQTT (boost_60_min) is received and the CAN message it is supposed to send is showing in the serial monitor, I thought perhaps the TX wire is disconnected between the ESP32 and CAN transceiver. Took the multi-meter to validate, pin I04 from the ESP module on the wemos has continuity all the way to the CAN transceiver.

I'd first like to see the CAN messages show up in the candump output on my rpi. The final step is to get the Q350 do something with it.

Perhaps a bit off-topic, but this may be of interest: https://www.international.zehnder-systems.com/connected-products

Marchellos, 2022-01-02:
> @asmanea I've uploaded simple connection diagram so it is easy even for a beginner. Keep in mind (as you can see in the photo) I DID NOT USED this exact diagram because I hate destroying ethernet [...]

### First of all, best wishes for the coming year!

First of all, best wishes for the coming year.

It's probably the wrong place to ask this, but I have the following question.

But is [the EW11](http://www.hi-flying.com/elfin-ew10-elfin-ew11?search=EW11) a ready-made solution?

Or do I have to have another model if I don't want to make it myself?
I'm not that good at doing stuff like this myself.

Uriziel01, 2022-01-02:
It's a pretty straightforward build on a tight budget, this one seems like it costs $24 on itself?

It would most propably work but it is just a wireless serial bridge and would require writing some additional software to handle this in HA or somewhere else.

Marchellos, 2022-01-02:
Thank you for your quick response.

The price of the Elfin-EW11 ranges from €10 - €20.
This depends on the additional items (antenna, cables, suspension bracket).
There is also a LAN version (Elfin-EE11).
For me the WIFI version would be a good solution; or does this make it very complicated?

Uriziel01, 2022-01-02:
The WiFi/eth versions does not change anyting implementation-wise as far as I know.

piwko28, 2022-01-03:
Is option box needed for the wiring?
My Waveshare SN65HVD230 is getting really hot when I attach it to CAN. Voltages are 1.5 between CAN pins and GND.

Uriziel01, 2022-01-04:
I do not have any option/KNX/LAN box installed, just a can transceiver and it is definitely not getting hot.

piwko28, 2022-01-04:
Thanks for your answer, it's important hint.
I meant voltage between CAN_L and GND and CAN_H and GND. Red connector has 12V as well.
I'll try with another can transceiver then, maybe the one is broken.

piwko28, 2022-01-08:
I had wrong wiring and tried to use espcam. It works perfectly with esp32.

pistacjusz, 2022-02-22:
After many tries, I finally got the project started. The cause of the problems was the factory broken can transciver. I changed it to another one and everything started working immediately, I can read the recuperator states and control it. Greetings:)

quentindavid, 2022-08-26:
Hi, thanks a lot for this project and this topic which are really helpful !
Something I don't quite understand : why would you use the ethernet cable if you can just plug CAN_H, CAN_L, +24V and GND with regular cables in the connector on the left of the RJ45 ?
Is it fine if I connect my CAN transceiver using this port instead of RJ45 ?

Uriziel01, 2022-08-26:
Sure you can use the terminals, the RJ45 is just more convenient, especially when setting things up I had few times some issue and the only way to resolve it was to just physically unplug the RJ45 cable as it was preventing the whole unit from booting up properly.

pyzimmer, 2022-10-18:
I think you can wire the +12v directly to the VIN pin of the ESP32, because ESP board have a regulator. You don't need DC-DC converter.

Uriziel01, 2022-10-18:
@pyzimmer I do not recommend to hook up 12V directly to ESP32 board, it does in fact have a regulator but it is a linear regulator not a switching one, if regulating down from 12V to 5V it will cause regulator to dissipate lots of heat.

At around 160mA-250mA average power draw for ESP32 and 7V potential difference (12-5) it's around 1.12W to 1.75W of power that will get converter into heat. It's fine for a short period but in this case device will run 24/7.

Hans-0, 2023-01-10:
For this project I am using the D1 Mini ESP32 ESP-32 Wifi + Bluetooth (CP2102) - see image:

[image]

[image]

Vekexasia says in his description:
can_rx -> pin 5 of esp32
can_tx -> pin 4 of esp32

Which pins are those exactly on my esp32?

Uriziel01, 2023-01-10:
**D4** is the **GPIO4** (**IO4** in your case)
**D5** is the **GPIO5** (**IO5** in your case)

[image]

rpiLovers, 2023-06-21:
Hi, thanks a lot for this project and this topic!
I just need some help with installation (ESPHOME newbie....)
I have WT32-ETH01 ESP32  board, Waveshare SN65HVD230 CAN Tranceiver Module and Buck Converter LM2596 3A.
In home assistant I see the ESP board and all of the parameters (i used the external_components:   - source: github://vekexasia/comfoair-esp32).
The problem is that I am not sure which ports to use and to change in the code.
I think RXD (GPIO05) and TXD (GPIO17).
How can I change the code in ESPHOME? Your board uses PIn 4 and 5.
Hope someone can helpout.
I would love to get my Q450 working on the network.
Thanks in advance.
HDR

pat1300, 2023-06-21:
> @asmanea I've uploaded simple connection diagram so it is easy even for a beginner. Keep in mind (as you can see in the photo) I DID NOT USED this exact diagram because I hate destroying ethernet [...]

@Uriziel01 : Hi, are you sure of your connection diagram? CAN_L (white wire) is connected to CANH of the transceiver , and CAN_H ( yellow) is conncetd to CANL . I think it should be the reverse.  Your photos are correct, but not the diagram I think. 
Best regards,
Patrick

Uriziel01, 2023-06-22:
@pat1300 Just checked and I'm sure of the fact that it is definetly wrong 😄 **Of course CAN_L should be connected to CANL and CAN_H to CANH.**

@rpiLovers I guess you can just change 25/21 pins combo in the `comfoair-esp32/components/comfoair
/__init__.py` file. @vekexasia mentioned that he will be possibly adding custom pins configuration in the near future here https://github.com/vekexasia/comfoair-esp32/issues/46#issuecomment-1533125530

poingZ, 2023-06-27:
> @Uriziel01 : Hi, are you sure of your connection diagram? CAN_L (white wire) is connected to CANH of the transceiver , and CAN_H ( yellow) is conncetd to CANL . I think it should be the reverse. [...]

@pat1300 are you powering the ESP32 with the 3.3V DC-DC convertor? My setup doesn't work and can't find out why, I'm powering the ESP32 (mini) via USB and the ESP32 is giving 3.3v to the CAN module (I measured it). I don't see why this shouldn't work ...

vekexasia, 2024-03-11:
Hello @Uriziel01 @pat1300 @rpiLovers You can now do it via rx_pin and tx_pin configuration flags

closed 2024-03-11 by vekexasia