vekexasia/comfoair-esp32 #6: bypass_state discussion

svde, 2021-07-10:
Not really an issue, but more something I'd like to discuss.

I noticed that bypass_state is treated like on/off. I reality the bypass valve can be opened partially. I'm also seeing that from candump:

  can0  0038C041   [1]  64
  can0  0038C041   [1]  5D
  can0  0038C041   [1]  64
  can0  0038C041   [1]  00
  can0  0038C041   [1]  64
  can0  0038C041   [1]  00
  can0  0038C041   [1]  64
  can0  0038C041   [1]  00

Would it be possible to implement it as a percentage instead?

-      LAZYSWITCH(227, "bypass_state", "%s", vals[0] == 64 ? "on" : "off")  // 1 = on, 0 = off   
+      LAZYSWITCH(227, "bypass_state", "%d", vals[0])  // %

vekexasia, 2021-07-10:
hey @svde why not. i guess so. anyhow i still have to understand what is the bypass_state exactly.

svde, 2021-07-10:
OK, it's included in #7

closed 2021-07-10 by svde