vekexasia/comfoair-esp32 #14: esp keeps losing wifi connection

tobiaranz, 2022-03-18:
I had this issue that the esp would keep losing wifi connection until it was restarted manually. Wifi connection only lasted 1-2 of hours before it dropped.
Despite the ::WiFi.setAutoReconnect(true); function in wifi.cpp, it would not reconnect.

So I found a workaround, adding this simple function inside the void WiFi::loop

if (::WiFi.status() != WL_CONNECTED) {
      ::WiFi.begin(WIFI_SSID, WIFI_PASS);
      }

Since this update I have not had the problem anymore
Anybody had the same issue?

vekexasia, 2022-04-12:
Hey did you have the same behavior as the one expressed in https://github.com/vekexasia/comfoair-esp32/issues/17 ?

tobiaranz, 2022-04-12:
Yes, I have a similar behaviour, but I can't figure out if it is due to the esp board or the comfoair.
I tried adding a wifi relay in order to restart the whole  VMC when it doesn't send any new data for ~5 minutes. This works some times, but also happens that giving back power with the esp/can connected, the VMC doesn't boot up.

The code I added anyway prevents the esp from losing the wifi connection, which used to happen often even if the signal is good in that area. 

I'm using an ESP32-WROOM-32 (D1 mini) and the CAN transceiver SN65HVD230