vekexasia/comfoair-esp32 #16: ventilation_extract_only for more than 1 h

zumek66, 2022-03-23:
Hi
I need to have ventilation_extract_only for 12 hours instead of 1 hour active. I am not a programmer but after some research it seems like this part of code needs to be modified in commands.h:

#define CMD_ventilation_extract_only        { 0x84, 0x15, 0x07, 0x01, 0x00, 0x00, 0x00, 0x00, **0x10, 0x0e**, 0x00, 0x00, 0x00 }

new value:
#define CMD_ventilation_extract_only        { 0x84, 0x15, 0x07, 0x01, 0x00, 0x00, 0x00, 0x00, **0xc0, 0xa8**, 0x00, 0x00, 0x00 }

Is it correct? New value is correct?

Thank You

zumek66, 2023-03-24:
I've prepared values for 3,6,12 h. Maybe this will be useful for someone.

0x84, 0x15, 0x07, 0x01, 0x00, 0x00, 0x00, 0x00, 0x10, 0x0e, 0x00, 0x00, 0x00  - 1h
0x84, 0x15, 0x07, 0x01, 0x00, 0x00, 0x00, 0x00, 0x30, 0x2A, 0x00, 0x00, 0x00  - 3h
0x84, 0x15, 0x07, 0x01, 0x00, 0x00, 0x00, 0x00, 0x60, 0x54, 0x00, 0x00, 0x00  - 6h
0x84, 0x15, 0x07, 0x01, 0x00, 0x00, 0x00, 0x00, 0xC0, 0xA8, 0x00, 0x00, 0x00 - 12h

I've tested 3h value change. Working as expected.

vekexasia, 2023-04-28:
#46 will allow custom hex sending.

vekexasia, 2023-04-28:
leaving here for reference:

service: esphome.comfoair_send_hex
data:
  hexSequence: "8415070100000000100e000000"