vekexasia/comfoair-esp32 #19: assert failed: xTaskRemoveFromEventList tasks.c

pyto00, 2022-05-12:
I am struggeling with this assert fail error. I found that this is a part of FreeRTOS which I have no knowledge about.
It reboots a few times and then gets stuck.

From the monitor terminal

```
Connecting to INET
.......
WiFi Connected. IP:
192.168.0.45
ESP_SMT init
ESPVMC init
Attempting MQTT connection...connected
Subscribing to: comfoair/commands/auto
Subscribing to: comfoair/commands/boost_10_min

assert failed: xTaskRemoveFromEventList tasks.c:3657 (pxUnblockedTCB)

Backtrace:0x40083835:0x3ffb23400x4008b7e9:0x3ffb2360 0x40090ca9:0x3ffb2380 0x4008d586:0x3ffb24b0 0x4008c8b8:0x3ffb24d0 0x40100c32:0x3ffb2510 0x40100f61:0x3ffb2530 0x400efb26:0x3ffb2550 0x400f14b8:0x3ffb2580 0x400ead15:0x3ffb2620 0x400d8522:0x3ffb2690 0x400d7ab0:0x3ffb26f0 0x400d7e1d:0x3ffb2710 0x400d7e35:0x3ffb2730 0x400d4adb:0x3ffb2750 0x400d4ba3:0x3ffb2790 0x400d4c1e:0x3ffb27e0 0x400d49bc:0x3ffb2800 0x400e0b51:0x3ffb2820 
```

Does someone has any pointers for me ?

svde, 2022-05-12:
Same as in my (updated) [comment](https://github.com/vekexasia/comfoair-esp32/issues/13#issuecomment-1123989770) on a previous issue.

pyto00, 2022-05-13:
The code written by @vekexasia is not compatible anymore with the latest ESPRESSIF32 platform.
I went to this overview of all the versions to see what version they where at now.

[https://registry.platformio.org/platforms/platformio/espressif32/versions?version=3.3.0](url)

I then randomly chose a version from 10 months ago.
and edited the platformio.ini config like below.

```
[env:nodemcu-32s]`
platform = platformio/espressif32@3.3.0
board = esp32dev
framework = arduino
monitor_speed = 115200
upload_port = COM20
monitor_port = COM20
lib_deps = 
    SPI
	can_common=https://github.com/collin80/can_common.git
	esp32_can=https://github.com/collin80/esp32_can.git
	https://github.com/knolleary/pubsubclient.git
```

the ESP32 does not crash anymore and I was able to send a command.
```
Connecting to INET
.........
WiFi Connected. IP:
192.168.0.45
ESP_SMT init
ESPVMC init
Attempting MQTT connection...connected
Subscribing to: comfoair/commands/auto
Subscribing to: comfoair/commands/boost_10_min
Subscribing to: comfoair/commands/boost_20_min
Subscribing to: comfoair/commands/boost_30_min
Subscribing to: comfoair/commands/boost_60_min
Subscribing to: comfoair/commands/boost_end
Subscribing to: comfoair/commands/bypass_activate_1h
Subscribing to: comfoair/commands/bypass_auto
Subscribing to: comfoair/commands/bypass_deactivate_1h
Subscribing to: comfoair/commands/manual
Subscribing to: comfoair/commands/set_mode
Subscribing to: comfoair/commands/temp_profile_cool
Subscribing to: comfoair/commands/temp_profile_normal
Subscribing to: comfoair/commands/temp_profile_warm
Subscribing to: comfoair/commands/ventilation_extract_only
Subscribing to: comfoair/commands/ventilation_extract_only_reset
Subscribing to: comfoair/commands/ventilation_level
Subscribing to: comfoair/commands/ventilation_level_0
Subscribing to: comfoair/commands/ventilation_level_1
Subscribing to: comfoair/commands/ventilation_level_2
Subscribing to: comfoair/commands/ventilation_level_3
Subscribing to: comfoair/commands/ventilation_supply_only
Subscribing to: comfoair/commands/ventilation_supply_only_reset
-------new message from broker-----
channel:comfoair/commands/ventilation_level_3
data:
Received: ventilation_level_3
1F034051 00 84 15 01 01 00 00 00
1F034051 81 00 01 00 00 00 03
1F054051 00 84 15 01 01 00 00 00 
1F054051 81 00 01 00 00 00 03
```

vekexasia, 2022-05-13:
hey @pyto00 can you please open a merge request?

piotrsn, 2022-06-17:
> @svde Tnx for your input! The code written by @vekexasia is not compatible anymore with the latest ESPRESSIF32 platform. I went to this overview of all the versions to see what version they where at now.
> 
> [https://registry.platformio.org/platforms/platformio/espressif32/versions?version=3.3.0](url)
> 
> I then randomly chose a version from 10 months ago. and edited the platformio.ini config like below.
> 
> ```
> [env:nodemcu-32s]`
> platform = platformio/espressif32@3.3.0
> board = esp32dev
> framework = arduino
> monitor_speed = 115200
> upload_port = COM20
> monitor_port = COM20
> lib_deps = 
>     SPI
> 	can_common=https://github.com/collin80/can_common.git
> 	esp32_can=https://github.com/collin80/esp32_can.git
> 	https://github.com/knolleary/pubsubclient.git
> ```
> 
> the ESP32 does not crash anymore and I was able to send a command.
> 
> ```
> Connecting to INET
> .........
> WiFi Connected. IP:
> 192.168.0.45
> ESP_SMT init
> ESPVMC init
> Attempting MQTT connection...connected
> Subscribing to: comfoair/commands/auto
> Subscribing to: comfoair/commands/boost_10_min
> Subscribing to: comfoair/commands/boost_20_min
> Subscribing to: comfoair/commands/boost_30_min
> Subscribing to: comfoair/commands/boost_60_min
> Subscribing to: comfoair/commands/boost_end
> Subscribing to: comfoair/commands/bypass_activate_1h
> Subscribing to: comfoair/commands/bypass_auto
> Subscribing to: comfoair/commands/bypass_deactivate_1h
> Subscribing to: comfoair/commands/manual
> Subscribing to: comfoair/commands/set_mode
> Subscribing to: comfoair/commands/temp_profile_cool
> Subscribing to: comfoair/commands/temp_profile_normal
> Subscribing to: comfoair/commands/temp_profile_warm
> Subscribing to: comfoair/commands/ventilation_extract_only
> Subscribing to: comfoair/commands/ventilation_extract_only_reset
> Subscribing to: comfoair/commands/ventilation_level
> Subscribing to: comfoair/commands/ventilation_level_0
> Subscribing to: comfoair/commands/ventilation_level_1
> Subscribing to: comfoair/commands/ventilation_level_2
> Subscribing to: comfoair/commands/ventilation_level_3
> Subscribing to: comfoair/commands/ventilation_supply_only
> Subscribing to: comfoair/commands/ventilation_supply_only_reset
> -------new message from broker-----
> channel:comfoair/commands/ventilation_level_3
> data:
> Received: ventilation_level_3
> 1F034051 00 84 15 01 01 00 00 00
> 1F034051 81 00 01 00 00 00 03
> 1F054051 00 84 15 01 01 00 00 00 
> 1F054051 81 00 01 00 00 00 03
> ```

This helped me a lot - thank you. The script is now fully working.

vekexasia, 2022-06-17:
Heyhey @piotrsn can you create a MR?

piotrsn, 2022-06-17:
Hey, to be honest I am soo new to this that I don't know how to do it. If
you can give me a guide or a link how to do so, then I will do. Thank you

atomic339, 2022-06-27:
If I want to use nodemcu 32S bord do I need to add:
[env:nodemcu-32s] in platformio.ini?

pyto00, 2022-07-03:
> hey @pyto00 can you please open a merge request?

Hi,

The merge request is still pending. Can you initiate the merge ?

vekexasia, 2022-07-03:
sorry pyto didn't see the MR for some reason. I just merged.

-- 

Andrea Baccega
[image: ***@***.***
[image: ***@***.***
[image: Personal site] <http://www.andreabaccega.com/> [image: Linkedin
Profile] <http://it.linkedin.com/in/andreabaccega> [image: Facebook]
<https://www.facebook.com/andrea.baccega> [image: Google+]
<https://plus.google.com/109217393200753135791> [image: Twitter]
<http://twitter.com/veke87> [image: StackOverflow]
<http://stackoverflow.com/users/314407/andrea-baccega> [image: GitHub]
<https://github.com/vekexasia>

closed 2022-09-12 by vekexasia