vekexasia/comfoair-esp32 #21: commands not accepted by comfoair unit

pyto00, 2022-05-13:
When the ESP32 is connected to the canbus I get an error on de display of the unit itself
`Comfonet error Please check cables`

I can only reset this error when I disconnect the ESP32.
While connected I am able to read out the sensor data but the unit doesn't react on commands.

Comfoair Firmware 1.9.0

Can this be a canbus ID error ? Maybe due to the recent firmare on the comfoair unit ?
[image]

panosnl, 2022-07-22:
I can send commands without errors to the Q600 with firmware 1.9.1 

I only get this error when the eps restarts.

masafra86, 2022-10-01:
I have the same problem, I can't send commands. I receive all the correct data (even if 15/20 minutes after resetting the ESP, connection fails), but I am not able to send commands. Firmware version R1.9.0

masafra86, 2022-11-01:
Today I got again to work on my ESP32 to try to understand the problem, and monitoring the ESP while I was trying to send any command, I found that it never receives any command message. I got back to build the firmware and I found in the logs this:

mpiling .pio/build/az-delivery-devkit-v4/src/main.cpp.o
Compiling .pio/build/az-delivery-devkit-v4/src/mqtt/mqtt.cpp.o
src/comfoair/comfoair.cpp: In member function 'void comfoair::ComfoAir::setup()':
src/comfoair/comfoair.cpp:24:4: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   });
    ^
src/comfoair/comfoair.cpp:42:5: note: in expansion of macro 'subscribe'
     subscribe("ventilation_level_0");
     ^
src/comfoair/comfoair.cpp:24:4: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   });
    ^
src/comfoair/comfoair.cpp:43:5: note: in expansion of macro 'subscribe'
     subscribe("ventilation_level_1");
     ^
src/comfoair/comfoair.cpp:24:4: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   });
    ^
src/comfoair/comfoair.cpp:44:5: note: in expansion of macro 'subscribe'
     subscribe("ventilation_level_2");
     ^
src/comfoair/comfoair.cpp:24:4: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   });
    ^
src/comfoair/comfoair.cpp:45:5: note: in expansion of macro 'subscribe'
     subscribe("ventilation_level_3");
     ^
src/comfoair/comfoair.cpp:24:4: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   });
    ^
src/comfoair/comfoair.cpp:46:5: note: in expansion of macro 'subscribe'
     subscribe("boost_10_min");
     ^
src/comfoair/comfoair.cpp:24:4: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   });
    ^
src/comfoair/comfoair.cpp:47:5: note: in expansion of macro 'subscribe'
     subscribe("boost_20_min");
     ^
src/comfoair/comfoair.cpp:24:4: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   });
    ^
src/comfoair/comfoair.cpp:48:5: note: in expansion of macro 'subscribe'
     subscribe("boost_30_min");
     ^
src/comfoair/comfoair.cpp:24:4: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   });
    ^
src/comfoair/comfoair.cpp:49:5: note: in expansion of macro 'subscribe'
     subscribe("boost_60_min");
     ^
src/comfoair/comfoair.cpp:24:4: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   });
    ^
src/comfoair/comfoair.cpp:50:5: note: in expansion of macro 'subscribe'
     subscribe("boost_end");
     ^
src/comfoair/comfoair.cpp:24:4: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   });
    ^
src/comfoair/comfoair.cpp:51:5: note: in expansion of macro 'subscribe'
     subscribe("auto");
     ^
src/comfoair/comfoair.cpp:24:4: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   });
    ^
src/comfoair/comfoair.cpp:52:5: note: in expansion of macro 'subscribe'
     subscribe("manual");
     ^
src/comfoair/comfoair.cpp:24:4: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   });
    ^
src/comfoair/comfoair.cpp:53:5: note: in expansion of macro 'subscribe'
     subscribe("bypass_activate_1h");
     ^
src/comfoair/comfoair.cpp:24:4: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   });
    ^
src/comfoair/comfoair.cpp:54:5: note: in expansion of macro 'subscribe'
     subscribe("bypass_deactivate_1h");
     ^
src/comfoair/comfoair.cpp:24:4: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   });
    ^
src/comfoair/comfoair.cpp:55:5: note: in expansion of macro 'subscribe'
     subscribe("bypass_auto");
     ^
src/comfoair/comfoair.cpp:24:4: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   });
    ^
src/comfoair/comfoair.cpp:56:5: note: in expansion of macro 'subscribe'
     subscribe("ventilation_supply_only");
     ^
src/comfoair/comfoair.cpp:24:4: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   });
    ^
src/comfoair/comfoair.cpp:57:5: note: in expansion of macro 'subscribe'
     subscribe("ventilation_supply_only_reset");
     ^
src/comfoair/comfoair.cpp:24:4: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   });
    ^
src/comfoair/comfoair.cpp:58:5: note: in expansion of macro 'subscribe'
     subscribe("ventilation_extract_only");
     ^
src/comfoair/comfoair.cpp:24:4: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   });
    ^
src/comfoair/comfoair.cpp:59:5: note: in expansion of macro 'subscribe'
     subscribe("ventilation_extract_only_reset");
     ^
src/comfoair/comfoair.cpp:24:4: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   });
    ^
src/comfoair/comfoair.cpp:60:5: note: in expansion of macro 'subscribe'
     subscribe("temp_profile_normal");
     ^
src/comfoair/comfoair.cpp:24:4: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   });
    ^
src/comfoair/comfoair.cpp:61:5: note: in expansion of macro 'subscribe'
     subscribe("temp_profile_cool");
     ^
src/comfoair/comfoair.cpp:24:4: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   });
    ^
src/comfoair/comfoair.cpp:62:5: note: in expansion of macro 'subscribe'
     subscribe("temp_profile_warm");
     ^
src/comfoair/comfoair.cpp:70:6: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     });
      ^
src/comfoair/comfoair.cpp:80:6: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     });
      ^

I'm not able to understand what is he saying, but maybe someone has the solution

closed 2024-03-11 by vekexasia