vekexasia/comfoair-esp32 #22: How to install haconfig files?

panosnl, 2022-07-13:
Hi,

I have the comfoair to mqtt running but now I have problems getting the haconfig file properly integrated in HA.

I tried !include index.yaml in the configuration.yaml but that seemed now to work. It does not like a custom component either or is it?

closed 2022-07-22 by panosnl

J1Buskens, 2022-09-30:
How did you fix it?

panosnl, 2022-10-02:
> How did you fix it?

I was new to HA so the 'manual' on github was not clear to me and also syntax seemed outdated. 
Eventually i got it working using the following settings

in configuration.yaml:

```
mqtt: !include mqtt.yaml
switch:
- platform: template
  switches:
    vmc_supply_ventilation:
      value_template: "{{ states('sensor.vmc_supply_fan_duty') | int > 0 }}"
      turn_on:
        service: mqtt.publish
        data:
          topic: comfoair/commands/ventilation_extract_only_reset
          payload: ':)'
      turn_off:
        service: mqtt.publish
        data:
          topic: comfoair/commands/ventilation_extract_only
          payload: ':)'

    vmc_extract_ventilation:
      value_template: "{{ states('sensor.vmc_exhaust_fan_duty') | int > 0 }}"
      turn_on:
        service: mqtt.publish
        data:
          topic: comfoair/commands/ventilation_supply_only_reset
          payload: ':)'
      turn_off:
        service: mqtt.publish
        data:
          topic: comfoair/commands/ventilation_supply_only
          payload: ':)'
```

in automations.yaml:
```
- id: Comfoair1
  alias: Comfoair nacht op 1
  description: ''
  trigger:
  - at: 00:00:00
    platform: time
  condition: []
  action:
  - service: mqtt.publish
    data:
      topic: comfoair/commands/ventilation_level_1
      payload: true
- id: Comfoair2
  alias: '[VMC] Auto Update user fan speed from MQTT'
  trigger:
    platform: mqtt
    topic: comfoair/fan_speed
  action:
  - service: input_select.select_option
    data:
      entity_id: input_select.vmc_user_fan_speed
      option: '{{ trigger.payload }}'
- id: Comfoair3
  alias: '[VMC] Set user fan speed on VMC from input select'
  trigger:
  - platform: state
    entity_id:
    - input_select.vmc_user_fan_speed
  action:
[... 109 lines cut ...]
      state_topic: "comfoair/exhaust_air_humidity"
      unit_of_measurement: "%"
    - name: "VMC Extract Air Humidity"
      unique_id: "VMC Extract Air Humidity"
      state_topic: "comfoair/extract_air_humidity"
      unit_of_measurement: "%"
    - name: "VMC Supply Air Humidity"
      unique_id: "VMC Supply Air Humidity"
      state_topic: "comfoair/supply_air_humidity"
      unit_of_measurement: "%"
```

Note that fan parameters are updated every second making my HA database very slow so I excluded these from the database

```
recorder:
  exclude:
    entities:
      - sensor.vmc_exhaust_fan_duty
      - sensor.vmc_exhaust_fan_flow
      - sensor.vmc_exhaust_fan_speed
      - sensor.vmc_supply_fan_duty
      - sensor.vmc_supply_fan_flow
      - sensor.vmc_supply_fan_speed
```

Obelixje, 2022-11-11:
Hi PanosNL,

Thank you for the shared yaml files. This did help me set it up.
I have some questions for you, I also am just starting with HomeAssistant and MTQQ, so newbe still...

Did you still use the index.html?

Can you maybe share the yaml file from your dashboard? Maybe this helps me understand how to set it up with the automation and swtiches.

Regards,
Rein

Obelixje, 2022-11-22:
Hi Panos,

I am trying to get the Comfoair back to auto.
I am using your automation.
Does this work at your end?

What exactly is the paypoad you are using? (quotes?)

- id: Comfoair5
  alias: '[VMC] Set user operating mode on VMC from input select'
  trigger:
    platform: state
    entity_id: input_select.vmc_operating_mode
  action:
  - service: mqtt.publish
    data:
      topic: comfoair/commands/set_mode
      payload: '{{ states(''input_select.vmc_operating_mode'') }}'

panosnl, 2022-11-23:
@Obelixje I have seem to have missed uploading the 2 input select configurations. They are indeed needed for everthing to work correctly.

I missed them because in the new HA it is defined in the GUI instead of YAML.

[image]
[image]

Obelixje, 2022-11-23:
Thank you,

This helps.
Can you please share the configuration of the button you are using for vmc_operating_mode in your UI?

panosnl, 2022-11-23:
> Thank you,
> 
> This helps. Can you please share the configuration of the button you are using for vmc_operating_mode in your UI?

Just make the input select helper and add it to the dashboard. It will create a pull down for you in the UI.

Obelixje, 2022-11-25:
Hi Panosnl,

Again, thanks for your help.
This now works.

I only have issues not when I use my boost button from the VMC it self. 
I use button to boost for 10 minutes, but the automation is switching to manual , level three. And this is not turned off after 10 minutes.
Without the ESP project the screen of the VMC always told me boost with a countdown from 10 to 0. Then back to auto.
Now manual level three. and this does not change back.

Do you have the same issue?