vekexasia/comfoair-esp32 #23: 5 Temperatures show incorrect values

panosnl, 2022-07-20:
All values seem to come correctly from my Q600 into MQTT.  

However 5 temp values show strange behavour: They cycle simultanuously the same (incorect) values: 0, 128, ~600) the latter value changes over time.

[image]

vekexasia, 2022-07-20:
hey @panosnl Maybe they changed the protocol? looks like you've a diff metric (not temperature) at those locations.

panosnl, 2022-07-23:
Update:

I noticed that the bogus values are updated every second. However I do see correct values coming by at a 'normal' rate that is expected 10-20 sec.

This means that the values are sent correctly after all but incorrect values are injected somehow. 
@vekexasia  Could it be that the decode for these parameters is falsely triggered ?

vekexasia, 2022-07-23:
so to make it clear you get "invalid values" AND "valid values" ?

It could be a bug in my code for the firmware 1.9.1 that can't decode values. Can you isolate just one metric and the raw you get annotating them?

panosnl, 2022-07-23:
Yes. that is hove it seems to be .
outdoor temp is 24.1.  I put some outdoortemp loggings in bold.

also it seems that Message : 1450201 X 10 gives both 0 and 128 ....

Cleaned up debug (decoded line needed a LF):

Message :  360041 X 274 0
Decoded :  ac_actual - 1.16
Message :  1444041 X 10 
Message :  1448081 X 10
Decoded :  extract_air_temp - 0.0
Message :  144C101 X 10
Decoded :  exhaust_air_temp - 0.0
**Message :  1450201 X 10** 
**Decoded :  outdoor_air_temp - 0.0**
Message :  1454401 X 10
Decoded :  pre_heater_temp_after - 0.0
Message :  1458801 X 10
Decoded :  post_heater_temp_before - 0.0
Message :  145D001 X 10
Message :  1462001 X 10
Message :  1E8041 X 237 5 
Decoded :  supply_fan_speed - 1335
Message :  1E4041 X 27B 5 
Decoded :  exhaust_fan_speed - 1403
Message :  10000001 X 0
Message :  370041 X 2F1 0 
Decoded :  pre_heater_temp_before - 24.1
**Message :  450041 X 2F1 0**
**Decoded :  outdoor_air_temp - 24.1**
Message :  454041 X 2F1 0
Decoded :  pre_heater_temp_after - 24.1
Message :  1E8041 X 235 5 
Decoded :  supply_fan_speed - 1333
Message :  1E4041 X 278 5 
Decoded :  exhaust_fan_speed - 1400
Message :  1444041 X 10
Message :  1448081 X 10 
Decoded :  extract_air_temp - 128.0
Message :  144C101 X 10
Decoded :  exhaust_air_temp - 128.0
**Message :  1450201 X 10**
**Decoded :  outdoor_air_temp - 128.0**
Message :  1454401 X 10
Decoded :  pre_heater_temp_after - 128.0
Message :  1458801 X 10 
Decoded :  post_heater_temp_before - 128.0
Message :  145D001 X 10
Message :  1462001 X 10
Message :  1DC041 X 2A4 0 
Decoded :  exhaust_fan_flow - 164
Message :  1D8041 X 11E
Decoded :  supply_fan_duty - 30
Message :  360041 X 275 0
Decoded :  ac_actual - 1.17
Message :  1E8041 X 23F 5 
Decoded :  supply_fan_speed - 1343
Message :  1E4041 X 276 5 
Decoded :  exhaust_fan_speed - 1398
Message :  10000001 X 0
Message :  1444041 X 10 
Message :  1448081 X 10
Decoded :  extract_air_temp - 128.0
Message :  144C101 X 10
Decoded :  exhaust_air_temp - 128.0
Message :  1450201 X 10
Decoded :  outdoor_air_temp - 128.0
Message :  1454401 X 10
Decoded :  pre_heater_temp_after - 128.0
Message :  1458801 X 10
Decoded :  post_heater_temp_before - 128.0
Message :  145D001 X 10
Message :  1462001 X 10
Message :  1E0041 X 2A4 0 
Decoded :  supply_fan_flow - 164
Message :  4C8041 X 21B 0
Message :  360041 X 276 0
Decoded :  ac_actual - 1.18
Message :  1E8041 X 248 5 
Decoded :  supply_fan_speed - 1352
Message :  1E4041 X 275 5 
Decoded :  exhaust_fan_speed - 1397

vekexasia, 2022-07-23:
Message : 1450201 X 10
Decoded : outdoor_air_temp - 128.0
and
Message : 1450201 X 10
Decoded : outdoor_air_temp - 0.0

there is something wrong within the decoding then. Cause looks like same emsage provides 2 diff values. 

Can you give it a try and reset it to 0 before decoding?

panosnl, 2022-07-23:
> Can you give it a try and reset it to 0 before decoding?

How would I do that? I only have been logging serial.print() data so far...

Also I don't see a part in the logging that I would expect based on the code:
I see the messageID , X, and length (eg. DEC 10) but not: 

    Serial.print(message->data.byte[i], HEX);
    Serial.print(" ");

```
void printFrame2(CAN_FRAME *message)
{
  Serial.print("Message :  ");
  Serial.print(message->id, HEX);
  if (message->extended) Serial.print(" X ");
  else Serial.print(" S ");   
  Serial.print(message->length, DEC);
  for (int i = 0; i < message->length; i++) {
    Serial.print(message->data.byte[i], HEX);
    Serial.print(" ");
 }
  Serial.println();
}
```

masafra86, 2022-09-05:
I have the same identical problem. Those 5 temperatures get quite strange values (0, 128, 500). I don't know how to debug them. 
Additionally, I'm not able to send commands to the uniting I can't understand why

vekexasia, 2022-09-12:
Hey guys can you try to change line 136 of message.cpp to this?

    uint16_t PDOID = (frame->id & 0x01fff000) >> 14;
?
There is an added `1`

panosnl, 2022-09-12:
> Hey guys can you try to change line 136 of message.cpp to this?
> 
> ```
> uint16_t PDOID = (frame->id & 0x01fff000) >> 14;
> ```
> 
> ? There is an added `1`

Thanks for the fix....
No weird values anymore it seems.

[image]

masafra86, 2022-09-12:
I confirm for the values. Now, what about the commands? Do you have any fix to give?

vekexasia, 2022-09-12:
can you open a new ticket masafra please and link it here? so that anyone else having issues with command can relate.

closed 2022-09-12 by vekexasia