vekexasia/comfoair-esp32 #24: New configuration for mqtt 2022.12.0

atomic339, 2022-09-10:
[image]

Do you make a new configuration or the old one will work?

vekexasia, 2022-09-10:
I did change mine. I might commit a new version

atomic339, 2022-09-11:
If you can it will be great

atomic339, 2022-09-15:
Can you help me with this changes?
Put this to my config but sensors don't work:
mqtt:
  sensor:
    # Consumption
    - name: "VMC Power Current"
      state_topic: "comfoair/power_consumption_current"
      unit_of_measurement: W
    - name: "VMC Total Energy Saving"
      state_topic: "comfoair/ah_ytd"
      unit_of_measurement: Wh

# FAns
    - name: "VMC Exhaust Fan Speed"
      state_topic: "comfoair/exhaust_fan_speed"
      unit_of_measurement: rpm
    - name: "VMC Supply Fan Speed"
      state_topic: "comfoair/supply_fan_speed"
      unit_of_measurement: rpm
    - name: "VMC Exhaust Fan Duty"
      state_topic: "comfoair/exhaust_fan_duty"
      unit_of_measurement: rpm
    - name: "VMC Supply Fan Duty"
      state_topic: "comfoair/supply_fan_duty"
      unit_of_measurement: rpm
    - name: "VMC Supply Fan User Speed"
      state_topic: "comfoair/fan_speed"

  # FLOW
    - name: "VMC Exhaust   Fan Flow"
      state_topic: "comfoair/exhaust_fan_flow"
      unit_of_measurement: "m³/h"
    - name: "VMC Supply Fan Flow"
      state_topic: "comfoair/supply_fan_flow"
      unit_of_measurement: "m³/h"

# TEMPS
    - name: "VMC Outdoor Air Temperature"
      state_topic: "comfoair/outdoor_air_temp"
      unit_of_measurement: "°C"
    - name: "VMC Exhaust Air Temperature"
      state_topic: "comfoair/exhaust_air_temp"
      unit_of_measurement: "°C"
    - name: "VMC Extract Air Temperature"
      state_topic: "comfoair/extract_air_temp"
      unit_of_measurement: "°C"
    - name: "VMC Post Heater Temp Before"
      state_topic: "comfoair/post_heater_temp_before"
      unit_of_measurement: "°C"
    - name: "VMC Post Heater Temp After"
      state_topic: "comfoair/post_heater_temp_after"
      unit_of_measurement: "°C"

# Humidity
    - name: "VMC Outdoor Air Humidity"
      state_topic: "comfoair/outdoor_air_humidity"
      unit_of_measurement: "%"
    - name: "VMC Exhaust Air Humidity"
      state_topic: "comfoair/exhaust_air_humidity"
      unit_of_measurement: "%"
    - name: "VMC Extract Air Humidity"
      state_topic: "comfoair/extract_air_humidity"
      unit_of_measurement: "%"
    - name: "VMC Supply Air Humidity"
      state_topic: "comfoair/supply_air_humidity"
      unit_of_measurement: "%"

atomic339, 2022-09-15:
Ok, it was a problem with my board. Works good

closed 2022-09-15 by atomic339