vekexasia/comfoair-esp32 #29: Error uploading to ESP

J1Buskens, 2022-09-30:
When I try to upload the project to my ESP I get an error in the remoteesp32 part.
I get the following:
`Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = espota
Uploading .pio\build\remoteesp32\firmware.bin
13:30:08 [DEBUG]: Options: {'esp_ip': '192.168.1.126', 'host_ip': '0.0.0.0', 'esp_port': 3232, 'host_port': 46476, 'auth': '', 'image': '.pio\\build\\remoteesp32\\firmware.bin', 'spiffs': False, 'debug': True, 'progress': True, 'timeout': 10}
13:30:08 [INFO]: Starting on 0.0.0.0:46476
13:30:08 [INFO]: Upload size: 826672
Sending invitation to 192.168.1.126 ..........
13:31:48 [ERROR]: No response from the ESP
*** [upload] Error 1
======================================== [FAILED] Took 102.68 seconds ========================================

Environment      Status    Duration
---------------  --------  ------------
wemos_d1_mini32  SUCCESS   00:00:22.718
remoteesp32      FAILED    00:01:42.678`

When I check the serial monitor, MQTT is able to subscribe to all topics and I get an IP address, so wifi and MQTT seem to be working.

vekexasia, 2022-09-30:
Apparently ota is broken. but you successfully uploaded via usb :)

J1Buskens, 2022-09-30:
What does this mean? Will it work like that?

J1Buskens, 2022-09-30:
My bad, I thought I needed to upload both :)

closed 2022-09-30 by J1Buskens