vekexasia/comfoair-esp32 #31: The idea of doing a do-over on ESP32 + ETHERNET

kamil2466, 2022-10-02:
Hello, 
I make various pcb boards and I have a proposal to make a professional board for this project. I wanted to add MQTT broker control over ETHERNET, is it possible to rewrite the software under ETHERNET ?

vekexasia, 2022-10-02:
why not kamil :) everything is possible

-- 

Andrea Baccega
[image: ***@***.***
[image: ***@***.***
[image: Personal site] <http://www.andreabaccega.com/> [image: Linkedin
Profile] <http://it.linkedin.com/in/andreabaccega> [image: Facebook]
<https://www.facebook.com/andrea.baccega> [image: Google+]
<https://plus.google.com/109217393200753135791> [image: Twitter]
<http://twitter.com/veke87> [image: StackOverflow]
<http://stackoverflow.com/users/314407/andrea-baccega> [image: GitHub]
<https://github.com/vekexasia>

kamil2466, 2022-10-05:
Thank you for your reply :)
I will make the design of the board and make it available to everyone.
The controller will run on 12v DC power and Ethernet POE, and the outputs will be CAN bus, RS485 and 1-wire (ds18b20)

Obelixje, 2022-10-19:
Very interesting!

Vinzent82, 2022-12-08:
great @kamil2466!

kamil2466, 2022-12-08:
I made a PCB design and will order a prototype
[image]
[image]
[image]

Input power supply +12V or from USB_C (+5V). Available buses (CANBUS, MODBUS and 1-WIRE).

kamil2466, 2022-12-09:
https://docs.google.com/document/d/1q0n5UrU_Wb4M9KEMclKQVpAS4q2IFIikL5ap5B0-XQ0/edit?usp=sharing

tezet, 2022-12-16:
Looks great @kamil2466! I like the idea of making the board based on WT32 - ETH01. I was planning to design some boards for my SSR relays control in similar way. Are you going to publish the project files for the board design?

kamil2466, 2022-12-16:
> Looks great @kamil2466! I like the idea of making the board based on WT32 - ETH01. I was planning to design some boards for my SSR relays control in similar way. Are you going to publish the project files for the board design?

Yes they will be made available :) For now it is a prototype after launching and testing I will make it available to everyone
Go to boneio.eu we do home automation controller

Vinzent82, 2023-02-07:
anybody having an idea how the firmware needs to be adapted for Ethernet support?

tezet, 2023-02-13:
I would personally go with porting it as a ESPHome component, it would make it more flexible to integrate with different things (including Ethernet). Unfortunately I currently don't have time to do it, also current solution works fine for me.

gahujipo, 2023-02-14:
@tezet what is your current solution? Does that solution use Ethernet?

vekexasia, 2023-04-15:
i'm designing a pcb as well. why would you want to use ethernet instead of wifi?

Vinzent82, 2023-04-30:
> why would you want to use ethernet instead of wifi?

@vekexasia, my reasons:

- better stability as my WiFiy quality is bad in the cellar
- we care for human radiation exposure. Thus we switch off our WiFi as frequently as possible and it runs on minimum radiation power
- less energy consumption

vekexasia, 2023-04-30:
interesting Vincent. thanks for sharing.

the board i designed (check main repo now) is modular. You can design a hat
with Ethernet and disable wifi .... :)

-- 

Andrea Baccega
[image: ***@***.***
[image: ***@***.***
[image: Personal site] <http://www.andreabaccega.com/> [image: Linkedin
Profile] <http://it.linkedin.com/in/andreabaccega> [image: Facebook]
<https://www.facebook.com/andrea.baccega> [image: Google+]
<https://plus.google.com/109217393200753135791> [image: Twitter]
<http://twitter.com/veke87> [image: StackOverflow]
<http://stackoverflow.com/users/314407/andrea-baccega> [image: GitHub]
<https://github.com/vekexasia>

hcouplet, 2023-05-30:
you can also directly use the board from olimex. See https://github.com/vekexasia/comfoair-esp32/pull/51

nliaudat, 2023-08-08:
That should works : https://github.com/nliaudat/esp_canbus 10$/board at jlcpcb

une18, 2023-11-25:
@kamil2466 What is the status of this project? Did you get it working?

gekkekoe, 2024-03-21:
@vekexasia  would one of these 
https://docs.m5stack.com/en/atom/Atomic%20RS485%20Base + https://docs.m5stack.com/en/core/AtomS3%20Lite
work out of the box?  I think that it should work? Might be an easy solution.

svde, 2024-03-22:
This one [https://docs.m5stack.com/en/unit/Unit-Mini%20CAN](https://docs.m5stack.com/en/unit/Unit-Mini%20CAN) should be able to power the board from the can bus.

I have one. I'm trying to use it in combination with an atoms3 lite / atoms3u, but the ESP32S3 doesn't compile. Will get a atom lite (regular esp32) and try with that.

gekkekoe, 2024-03-22:
> This one https://docs.m5stack.com/en/unit/Unit-Mini%20CAN should be able to power the board from the can bus.
> 
> I have one. I'm trying to use it in combination with an atoms3 lite / atoms3u, but the ESP32S3 doesn't compile. Will get a atom lite (regular esp32) and try with that.

yeah saw that one, but  can't order it (yet) in the Netherlands. Did you change the board definition to build esphome? the s3 should be supported (s3 is also faster). have a look at an example esphome: https://github.com/fonske/Mitsubishi_procon/blob/main/esphome/boards/board-esp32S3.yaml

svde, 2024-03-22:
This is the config I'm using:

```
substitutions:
  board: esp32-s3-devkitc-1
  device_description: "wtw new"
  mac: ***redacted***
  platform: "ESP32"
  rx_pin: "1" # RX = GPIO_NUM_1
  tx_pin: "2" # TX = GPIO_NUM_2

api:

captive_portal:

comfoair:
- rx_pin: ${rx_pin}
  tx_pin: ${tx_pin}

esphome:
  comment: "${device_description}"
  name: esphome-${mac}
  platformio_options:
    board_build.flash_mode: dio

esp32:
  board: ${board}
  framework:
    type: arduino

external_components:
  - source: github://vekexasia/comfoair-esp32
    components: [ comfoair ]

logger:
  level: INFO

mqtt:
  broker: ***redacted***

ota:

switch:
  - platform: restart
    name: restart

text_sensor:
  - platform: version
    name: version
  - platform: wifi_info
    ip_address:
      name: ip
    ssid:
      name: ssid
    bssid:
      name: bssid

web_server:
  port: 80

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: esphome-${mac}
    password: !secret ap_password
```

and this is the (first) error:
```
Compiling .pioenvs/esphome-802aa8c75ec3/FrameworkArduino/esp32-hal-gpio.c.o
In file included from /config/.esphome/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/dport_reg.h:21,
                 from .piolibdeps/esphome-802aa8c75ec3/esp32_can/src/esp32_can_builtin_lowlevel.cpp:33:
.piolibdeps/esphome-802aa8c75ec3/esp32_can/src/esp32_can_builtin_lowlevel.cpp: In function 'int CAN_init()':
.piolibdeps/esphome-802aa8c75ec3/esp32_can/src/esp32_can_builtin_lowlevel.cpp:217:29: error: 'DPORT_PERIP_CLK_EN_REG' was not declared in this scope
     DPORT_SET_PERI_REG_MASK(DPORT_PERIP_CLK_EN_REG, DPORT_CAN_CLK_EN);
                             ^~~~~~~~~~~~~~~~~~~~~~
```

gekkekoe, 2024-03-22:
Something like this should work:
`esp32:
  board: m5stack-atoms3
  variant: esp32s3
  framework:
    type: esp-idf
    version: 5.1.2
    platform_version: 6.5.0`

Seems to be ESPHome, it does not support platformIO 6 yet.

`In file included from src/esphome/components/comfoair/comfoair.cpp:1:
src/esphome/components/comfoair/comfoair.h:2:10: fatal error: esp32_can.h: No such file or directory
` It probably moved or is re-named differently in new version.

gekkekoe, 2024-03-22:
seems that we need to use an updated lib dep of the esp32_can.
I tried to replace it : https://github.com/gekkekoe/comfoair-esp32/commit/c79ed143059842534bc83159b5d50534fab15e61
but it does not seem to work, I'm pretty new with esphome so probably doing something wrong

svde, 2024-03-22:
Got the M5stack atom lite. It works, powered via the 12V of the zehnder unit.

gekkekoe, 2024-03-22:
use:
```
external_components:
  - source: github://gekkekoe/comfoair-esp32
    components: [ comfoair ]
```

and configure as

```
esp32:
  board: m5stack-atoms3
  variant: esp32s3
  framework:
    type: arduino
    version: 2.0.14
    platform_version: 6.5.0
```

this builds :) no need to buy a regular. the S3 is faster

vekexasia, 2024-03-22:
Can you guys post a pic of Al these different builds?

svde, 2024-03-23:
This is my second "build". 
[image]

Components:

- https://shop.m5stack.com/products/atom-lite-esp32-development-kit
- https://shop.m5stack.com/products/mini-can-unit-tja1051t-3
- 4 +/- 15cm wires (from a UTP cable)

My first build (the original 3D printed box) isn't able to send commands to the ventilation unit. That's why I built the 2nd one, which also isn't able to send commands to the ventilation unit. Perhaps it's due to the [ComfoConnect LAN C](https://www.international.zehnder-systems.com/en/comfortable-indoor-ventilation/products/connectivity/interface-module/zehnder-comfoconnect-lan-c) and [option box](https://www.international.zehnder-systems.com/en/comfortable-indoor-ventilation/products/connectivity/interface-module/zehnder-option-box) that are also connected to the unit.

gekkekoe, 2024-03-23:
@vekexasia  I still have a soldered version, but want to replace it. Would like a CAN module that can also power an atom3. Preferably no cable between the atom and CAN module. this https://docs.m5stack.com/en/atom/atom_can would have been ideal.

SkullboyNL, 2024-03-24:
i made a Wifi and ethernet version 

[image]
[image]
[image]
[image]

JicayM, 2025-04-15:
> C'est ma deuxième "construction". [image]
> 
> Composants:
> 
> * https://shop.m5stack.com/products/atom-lite-esp32-development-kit
> * https://shop.m5stack.com/products/mini-can-unit-tja1051t-3
> * 4 fils +/- 15 cm (provenant d'un câble UTP)
> 
> Il est suffisamment petit pour être rangé sous le couvercle de câblage sur le dessus de l'unité.
> 
> Ma première construction (la boîte originale imprimée en 3D) ne parvient pas à envoyer de commandes au système de ventilation. C'est pourquoi j'ai construit la deuxième, qui, en fin de compte, ne parvient pas non plus à envoyer de commandes au système de ventilation. Cela est peut-être dû au [ComfoConnect LAN C](https://www.international.zehnder-systems.com/en/comfortable-indoor-ventilation/products/connectivity/interface-module/zehnder-comfoconnect-lan-c) et au [boîtier d'options](https://www.international.zehnder-systems.com/en/comfortable-indoor-ventilation/products/connectivity/interface-module/zehnder-option-box) également connectés au système.

Hi,

I d'ont understand why it doesn't work with my Zehnder comfoair q600.

Here my config in esphome 👍 

`esphome:
  name: vmc
  friendly_name: vmc

esp32:
  board: m5stack-atoms3
  variant: esp32s3
  framework:
    type: arduino
    version: 2.0.14
    platform_version: 6.5.0

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "mRTlhf5FMNXgPdz1NBugf7//nDhK8etupArmMza5Ydo="

ota:
  - platform: esphome
    password: "9d812e17e14abd12564a9f4877ff33f7"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Vmc Fallback Hotspot"
    password: "mC4RxMSvFQ8X"

captive_portal:

external_components:
  - source: github://gekkekoe/comfoair-esp32
    components: [ comfoair ]
    `

Here my logs : 

`INFO ESPHome 2025.3.3
INFO Reading configuration /config/esphome/vmc.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Starting log output from 192.168.0.238 using esphome API
INFO Successfully connected to vmc @ 192.168.0.238 in 0.081s
INFO Successful handshake with vmc @ 192.168.0.238 in 0.059s
[22:56:08][I][app:100]: ESPHome version 2025.3.3 compiled on Apr 15 2025, 22:40:54
[22:56:08][C][wifi:600]: WiFi:
[22:56:08][C][wifi:428]:   Local MAC: 48:CA:43:B6:8D:C4
[22:56:08][C][wifi:433]:   SSID: 'Mesters'[redacted]
[22:56:08][C][wifi:436]:   IP Address: 192.168.0.238
[22:56:08][C][wifi:440]:   BSSID: 98:52:4A:BD:22:00[redacted]
[22:56:08][C][wifi:441]:   Hostname: 'vmc'
[22:56:08][C][wifi:443]:   Signal strength: -51 dB ▂▄▆█
[22:56:08][C][wifi:447]:   Channel: 11
[22:56:08][C][wifi:448]:   Subnet: 255.255.255.0
[22:56:08][C][wifi:449]:   Gateway: 192.168.0.1
[22:56:08][C][wifi:450]:   DNS1: 62.197.111.140
[22:56:08][C][wifi:451]:   DNS2: 109.88.203.3
[22:56:08][C][logger:177]: Logger:
[22:56:08][C][logger:178]:   Max Level: DEBUG
[22:56:08][C][logger:179]:   Initial Level: DEBUG
[22:56:08][C][logger:181]:   Log Baud Rate: 115200
[22:56:08][C][logger:182]:   Hardware UART: USB_CDC
[22:56:08][C][captive_portal:089]: Captive Portal:
[22:56:08][C][mdns:116]: mDNS:
[22:56:08][C][mdns:117]:   Hostname: vmc
[22:56:08][C][esphome.ota:073]: Over-The-Air updates:
[22:56:08][C][esphome.ota:074]:   Address: vmc.local:3232
[22:56:08][C][esphome.ota:075]:   Version: 2
[22:56:08][C][esphome.ota:078]:   Password configured
[22:56:08][C][safe_mode:018]: Safe Mode:
[22:56:08][C][safe_mode:020]:   Boot considered successful after 60 seconds
[22:56:08][C][safe_mode:021]:   Invoke after 10 boot attempts
[22:56:08][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[22:56:08][C][api:140]: API Server:
[22:56:08][C][api:141]:   Address: vmc.local:6053
[22:56:08][C][api:143]:   Using noise encryption: YES
`

And pictures of my connections but I have nothing in Home assistant ... 

[image]
[image]
[image]

Can you help me ?

THank you very much for your help !

gekkekoe, 2025-04-16:
you probably need to use this board definition
```yaml
esp32:
  board: esp32-s3-devkitc-1
```

JicayM, 2025-04-16:
Thansk I will try !

When you say : The groove port uses GPIO 1 and 2

I have to put this code 

`comfoair:
  id: comfo
  rx_pin: GPIO1
  tx_pin: GPIO2`

And not : 

`comfoair:
  id: comfo
  rx_pin: GPIO5
  tx_pin: GPIO4`

That's right ?

gekkekoe, 2025-04-16:
yes but double check which pin is Rx/Tx

JicayM, 2025-04-16:
ok and i guess if it doesn't work just reverse the configuration

gekkekoe, 2025-04-16:
@JicayM did you double check the rx/tx ?
post that section

@svde mind to share your config for the atom ?

JicayM, 2025-04-16:
i found the problem ! thanks a lot