vekexasia/comfoair-esp32 #32: Project compilation problems

MichalAug, 2022-10-02:
Dear All,
I have a problem compiling a project. After many hours I have reached the stage that I am getting an error related to the missing secrets.h file. But when I add it in the src folder, I get a double variable declaration error. What am I doing wrong? What should it look like? Please help :)
[image]

MichalAug, 2022-10-02:
When I put this in secrets.h
const char* WIFI_SSID = "123";
const char* WIFI_PASS = "123";
const char* MQTT_USER = "mqttuser";
const char* MQTT_PASS = "mqtt pass";
const char* MQTT_PREFIX = "comfoair";
const char* MQTT_HOST = "192.168.2.100";
const char* MQTT_PORT = "6060";

I got:
src/comfoair/comfoair.cpp: In member function 'void comfoair::ComfoAir::setup()':
src/comfoair/comfoair.cpp:20:58: error: expected ')' before string constant
 #define subscribe(command) mqtt->subscribeTo(MQTT_PREFIX "/commands/" command, [this](char const * _1,uint8_t const * _2, int _3) { \

jakubko186, 2022-10-11:
did you fix it?

SkullboyNL, 2022-10-14:
Look at issue [#9](https://github.com/vekexasia/comfoair-esp32/issues/9)

vekexasia, 2023-04-28:
You can just use esphome component now :)

closed 2023-04-28 by vekexasia