vekexasia/comfoair-esp32 #33: Integration with existing UI for CA350

mat3u, 2022-10-17:
Hello,

I was tinkering over weekend to make working this firmware with existing Lovelace controls (eg. https://github.com/mweimerskirch/lovelace-hacomfoairmqtt). Result:

[image]
I can see temperatures, fan duty and can control fan level.

I made some changes, some of them makes sens (like last will) other probably could be handled by HA itself by smart configuration. Eg.:
```
fan_mode_command_topic: "comfoair/climate/fan/set"
fan_mode_state_topic: "comfoair/climate/fan"
fan_modes:
  - off
  - low
  - medium
  - high
 ```
Instead changing firmware these topics could be done using templates, but I was unable to write template that is working and produces results I want. 

You can see changes here: https://github.com/mat3u/comfoair-esp32/tree/hacomfoairmqtt-compatibility. If any of this features is interesting then, let me know.

vekexasia, 2022-10-17:
Hello @mat3u thanks for bringing to my attention that project. I was not aware of that. I've quickly gone over your changes and there are a lot that are basically reformatting the code.

If you could manage to remove the reformatting changes and keep compatibility to the current mqtt topics that would be a nice addition. Also for your 3d case which should allow more space than mine which does not leave much room.

mat3u, 2022-10-17:
Yeah, my editor reformated whatever I've touched, but current code is a little bit inconsistent ;) Maybe you have a preferred `.editorconfig` that I can use with VSCode to reformat it properly?

I can add old MQTT Topics without any problem and cleanup README.md a little bit. I'll work on that.

mat3u, 2022-10-17:
I reverted most of the auto-reformat things. I'll make some changes to make it backward-compatible.

I suggest, to reformat whole repository in single PR, as it will be hard to develop in this repository without it.

jakubko186, 2022-11-22:
Can you see animation of the level 0 (fan off) button? After clicking level 0 it works (sends correct mqtt command), but I still see in blue only the level I've used before :(
Levels 1, 2 and 3 works fine.

mat3u, 2022-11-26:
Yeah, it is also not working in my setup. I was digging yesterday into it, and it seems like HA is not respecting list of values to be parsed, and only passes default values. I'll try to debug it further.

mat3u, 2022-11-27:
Hey, I've been digging into it, and it seems that `off` is some keyword that has more meaning. My climate configuration looks like this rn: 
```
  fan_mode_command_topic: "comfoair/climate/fan/set"
  fan_mode_state_topic: "comfoair/climate/fan"
  fan_modes:
    - "off"
    - low
    - medium
    - high
  ```

You may notice that `off` is wrapped by `"`.
Also, Climate seem to be much more cooperative when I commented out whole `mode` section. It was generating a lot of errors and warnings.

closed 2024-07-04 by vekexasia