vekexasia/comfoair-esp32 #35: No MQTT

Obelixje, 2022-11-17:
Hi,

There is no MQTT traffic coming from the esp32.
Is there a way to check logging on the ESP32?

When connecting to the IP of the ESP32 directly I only see possibility to upload files.

mikka2, 2022-11-17:
You can check if ESP32 is sending data by connecting to MQTT broker that you set and check for incoming messages.

Obelixje, 2022-11-18:
There are no incoming messages on the broker.

Do we ONLY expect messages when there is data coming in from Comfoair?
I am asking this because I may have blew up the Can tranceiver.

Or do we expect messages always when ESP is on the wifi. (Even when comfoair is not yet connected)?

gsabbe, 2022-11-18:
In my experience messages are sent as they are received and decoded from
the comfoair.
Also the comfoair seems to only send data as it changes. So it took two
hours for all my variables to be set in Home Assistant.
One amelioration would be to send a status message every 60 seconds or so
just to say "I'm online".

Obelixje, 2022-11-18:
ok,
thanks.
My first impression was that the Can tranciever is defect. It could be the case that I have to wait some hours for the first message..
Thanks for your help'

gsabbe, 2022-11-18:
Things like fan speed would change rather quickly and are the first
messages I saw coming in. Can you connect your laptop to the ESP and open
the arduino console?
This way you'd see the message coming in, and perhaps triple check your
MQTT settings. (I managed to mess up the port and password when setting it
up the first time :))
I also blew up my ESP32 because I didn't tune the DC-DC step down to 3.3V
from the 12V set at the factory. The CAN module didn't care about the over
voltage thankfully.
Late night projects are always fun.

Obelixje, 2022-11-18:
LOL

Same thing happened to me, also blew up the ESP32 because of the 12v...
I did triple test the incoming voltage (+/-), but did not check the outcoming voltage. So on my first connection I blew up the esp32, and an errormessage of the Comfoair.

I am going to connect the esp32 to Arduino, and see if something happens.

gsabbe, 2022-11-18:
My comfoairq actually shutdown completely with my mistake, rather scary.
But it came back to life after unplugging/replugging the power.
The ESP 5V pin is a lot more tolerant and accepts unregulated 5-12V. But
the 3.3V wants regulated 3.3V and nothing else.

Obelixje, 2022-11-18:
Is is working, the schedule was not correct.
[image]

CanH and CANL were switched....

Obelixje, 2022-11-18:
@gsabbe 
Thanks for your help this morning?
Do you have a good dashboard (yaml) that you can share?

Obelixje, 2022-11-19:
All solved. Wiring issue....

closed 2022-11-19 by Obelixje