vekexasia/comfoair-esp32 #39: Wrong variable type (uint16) for pre_heater_temp_before and post_heater_temp_after

panosnl, 2022-12-10:
Today the temps dropped below zero and I noticed that one temp started showing wrong values. Since it was OK for T >0 I checked the code where it was declared as _uint16_

message.cpp, line 186:
`     LAZYSWITCH(220, "pre_heater_temp_before", "%.1f", uint16/10.0) // C°`
`     LAZYSWITCH(221, "post_heater_temp_after", "%.1f", uint16/10.0)  // C°`

Changing it to _int16_ should solve the problem (I have to wait for tonight to confirm..)

`     LAZYSWITCH(220, "pre_heater_temp_before", "%.1f", int16/10.0) // C°`
`     LAZYSWITCH(221, "post_heater_temp_after", "%.1f", int16/10.0)  // C°`

nebman, 2022-12-12:
Can confirm the problem over here with negative temps
```
$ mosquitto_sub -t "comfoair/pre_heater_temp_before" -v
comfoair/pre_heater_temp_before 6552.6
```

svde, 2022-12-17:
outdoor_air_temp and rmot are probably also affected.

vekexasia, 2023-04-28:
fixed in #46