vekexasia/comfoair-esp32 #40: How to connect

Obelixje, 2022-12-15:
Hi,

How can guide me to the best startup solution?
I am currently connecting to the Q450 using the RJ45 port.

- If I connect the project to the powered off Q450 and then power on the Q450, the Q450 does not start
- If I connect the project to a completely started Q450 there is an errormessage telling me there is something wrong with the cabling.
- If I connect directly after powering on sometimes it all works.

What is the best methode?
Is there a difference between using the RJ450 port or the white connectore?

bakermat, 2022-12-15:
Have a look at #1, the cabling schema on the main page for RJ45 seems to be incorrect. I ran into the same issue, I used [this one](https://github.com/vekexasia/comfoair-esp32/issues/1#issuecomment-871412942) and can confirm that works for the RJ45 connection.

Obelixje, 2022-12-16:
@bakermat 

I don't think cabling is the issue here.
The project did work. Only this week the supplier came by for maintenance on my heater. This guy also delivered the Q450.
I didn't want him to know that I hooked something up because of warranty. So I removed it.
After this I am unable to connect anymore.

I have still all setup un a breadboard, maybe one ore more of the connections are not good anymore.
I am going to check this today/tomorrow.

Thanks for your response.

But regarding powering up.
do you power up the Q, and then connect the ESP, or have it already connected and then power up?

bakermat, 2022-12-16:
You can connect the RJ45 when the Qx50 is powered up, no need to power down the unit first. If you're fiddling with the white ports and cabling them one by one from a breadboard then make sure to turn off the unit to be safe.

Obelixje, 2022-12-16:
Thanks,

I have just replaced the CAN.
Now I can power up the Q when the board is already connected. So probably CAN was broken.
Only not yet MQTT messages...

Obelixje, 2022-12-16:
Maybe also the ESP32 is broken? Or at least the ports..
There are no MQTT messages send.

Obelixje, 2022-12-17:
The red light on the esp does blink when it esp is on 3.3 v. no wifi, no mqtt
When I connect to 5 v there is no blinking, hooked on wifi , no mqtt

Probably esp is broken. Tonight a new one is delivered from Amazon. I will probably test tomorrow.

Obelixje, 2022-12-18:
Also with new esp32 no mqtt coming from the esp32.
The esp32 is online.

Is there a way to see the logging from the esp?

Obelixje, 2022-12-18:
When I try to reprogram the esp32 i see that the 2nd part is failing. (wifi)
Is this important? Because through wire already passes.

[image]

Obelixje, 2022-12-19:
Maybe my 2nd Can Transceiver is also broken?
If not it might be I blew up something in my Q450...

Obelixje, 2022-12-29:
New Can transceiver did solve the issue.
Up and running again!

closed 2022-12-29 by Obelixje