vekexasia/comfoair-esp32 #46: Project direction and new features

vekexasia, 2023-04-15:
After this project has been released it entered a sleep state. Project seemed mature enough to not get new updates.

Due to the currently opened number of issues which include #43 #45 #25 #26 as well as #32 and others. It looks to me that it was about time to invest some free time to get this project a little love.

Of course if you want to donate me some beer which is the fuel of developers I will always appreciate that.

Anyways. I'm working on getting this on a properly designed lowcost PCB, as soon as I get the PCB delivered and get to test it out i'll also migrate the project to ESPHOME (which someone mentioned on issue #31).

I ordered 15 pcbs, assuming the pcb design works I can ship others to anyone interested in getting it.

This is a preview of the PCB
[image]

It does measure 45x23mm which is pretty small. The only missing part is an M5Stack Atom Lite ESP32 board. I decided to design it that way cause I did want to keep cost low, and leverage modularity. If ESP32 dies you can always replace it with another M5Stack. 

Since the board is very simple and modular it could be designed so that it scales vertically. so, If someone wants some extra functionality such as an ethernet HAT, a design could be made to meet it that needs. 

Cheers

vekexasia, 2023-04-28:
project has been converted to esphome enjoy all :) IF you need a custom pcb i've some spare

closed 2023-04-28 by vekexasia

svde, 2023-05-01:
> project has been converted to esphome enjoy all :)

Excellent choice.

nebman, 2023-05-03:
Excellent, thank you very much!  
For anyone else upgrading their ESPs please notice, that the GPIO pins changed.

svde, 2023-05-03:
Would be nice to make the GPIO pins configurable.

vekexasia, 2023-05-03:
@svde yeah i thought about that too. But i think gpio 25 and 21 are almost always available in all boards so I didn't see much of a need letting the pin configurable.

vekexasia, 2023-05-03:
also for everyone who wants i can ship the board with components (not soldered)

svde, 2023-05-03:
People could also choose to DIY with an ESP8266, where pins 21 and 25 aren't available...

nebman, 2023-05-03:
@svde but ESP8266 doesn't have a CAN controller onboard, only ESP32

svde, 2023-05-03:
I thought that we just use GPIO pings on the ESP32 for RX & TX.

Not the built-in CAN controller on the ESP32.

vekexasia, 2023-05-03:
> I thought that we just use GPIO ping on the ESP32 for RX & TX.

Exactly.

The internal canbus from esp32 is not compatible as far as I know with
extended ids.

Btw svde i think adding custom pins to conf is pretty easy. Ill check if I
have time in the following days

wiljums, 2023-05-11:
> also for everyone who wants i can ship the board with components (not soldered)

What would be the cost for the board and shipment to the Netherlands?

vekexasia, 2023-05-11:
about 55€ shipped with all components.

-- 

Andrea Baccega
[image: ***@***.***
[image: ***@***.***
[image: Personal site] <http://www.andreabaccega.com/> [image: Linkedin
Profile] <http://it.linkedin.com/in/andreabaccega> [image: Facebook]
<https://www.facebook.com/andrea.baccega> [image: Google+]
<https://plus.google.com/109217393200753135791> [image: Twitter]
<http://twitter.com/veke87> [image: StackOverflow]
<http://stackoverflow.com/users/314407/andrea-baccega> [image: GitHub]
<https://github.com/vekexasia>

wiljums, 2023-05-11:
Is it also possible to ship it soldered?

vekexasia, 2023-05-22:
sorry for late reply @wiljums. Yes i do have one test board. It is not perfectly soldered so i wouldn't vouch for it. My suggestion would be to solder this yourself juist to make sure things are soldered properly. I'm not a good one :)

jhornych, 2023-07-06:
I guess there is CAN module available including the ESP32 from m5stack, which you are using anyway just for 22EUR, or am I wrong? [M5STACK CAN module](https://shop.m5stack.com/products/atom-canbus-kit-ca-is3050g)

fuatakgun, 2023-10-22:
Looking forward to hear more on this

zweefan, 2023-10-23:
I have soldered the unit for myself and created 4 extra units. I will sell the units. I you are interested please send me a direct message. RegardsStephan

wiljums, 2023-10-23:
Hi zweefan, I am interested and figuring out how to send you a direct message.