vekexasia/comfoair-esp32 #47: Bootloop after flashing the ESPHome version

SkullboyNL, 2023-05-05:
I am trying to flash the ESPHome version, but then the ESP enters a boot loop. I've tried different configurations, but it's still stuck in the bootloop.
The error i get after flashing:
`01:49:05]E (10278) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[01:49:05]E (10278) task_wdt:  - loopTask (CPU 1)
[01:49:05]E (10278) task_wdt: Tasks currently running:
[01:49:05]E (10278) task_wdt: CPU 0: IDLE
[01:49:05]E (10278) task_wdt: CPU 1: CAN_LORX
[01:49:05]E (10278) task_wdt: Aborting.
[01:49:05]
[01:49:05]abort() was called at PC 0x400ef275 on core 0
[01:49:05]
[01:49:05]
[01:49:05]Backtrace:0x400836b5:0x3ffbea2c |<-CORRUPTED
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400836b5: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402`

Current yaml:
`esphome:
  name: air-11

esp32:
  board: nodemcu-32s
  framework:
    type: arduino
    
# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "gfhgfhgshbhbdbvfdg3354dsgf"
  
wifi:
  ssid: "*****"
  password: "******"

external_components:
  - source: github://vekexasia/comfoair-esp32
    components: [ comfoair ]

comfoair:
`

Am I doing something wrong or am I missing another configuration line?

vekexasia, 2023-05-06:
hey, can you take a picture of wiring?

SkullboyNL, 2023-05-06:
I first tried without the can module, so only a separate ESP32. Then tried with the can module connected TX to GPIO21 and RX to GPIO25 not yet connected to the comfoair but still the same error message.

Does the ESP not start if the ESP does not receive a signal from the comfoair?

lskuba, 2023-05-19:
I have the same problem after I am adding below to my yaml:
`external_components:
  - source: github://vekexasia/comfoair-esp32
    components: [ comfoair ]

comfoair:`

TroLoos, 2023-05-21:
Same here, wanted to switch to ESPHome and it does go into a bootloop.

`[C][ota:469]: There have been 0 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
E (10347) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10347) task_wdt:  - loopTask (CPU 1)
E (10347) task_wdt: Tasks currently running:
E (10347) task_wdt: CPU 0: IDLE
E (10347) task_wdt: CPU 1: CAN_LORX
E (10347) task_wdt: Aborting.

abort() was called at PC 0x400f9a98 on core 0

Backtrace:0x400836dd:0x3ffbea2c |<-CORRUPTED`

My setup is: ESP32 DEVKIT V1 with SN65HVD230, CAN module connected TX -> G25, RX -> G21 (I also tried G4 and G5 from previous MQTT version - BTW, what should it be?).

My YAML is as follows:

`esphome:
  name: comfoairv2
  friendly_name: comfoairv2

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino

external_components:
  - source: github://vekexasia/comfoair-esp32
    components: [ comfoair ]

logger:
  level: DEBUG

api:
  encryption:
    key: "*REDACTED*"

ota:
  password: "*REDACTED*"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "Comfoairv2 Hotspot"
    password: "*REDACTED*"

captive_portal:

comfoair:`

vekexasia, 2023-05-21:
That is strange can you share how you wired the board and which board it is?

TroLoos, 2023-05-22:
> That is strange can you share how you wired the board and which board it is?
> […](#)

Well, I have exactly this board:

https://botland.com.pl/moduly-wifi-i-bt-esp32/8893-esp32-wifi-bt-42-platforma-z-modulem-esp-wroom-32-zgodny-z-esp32-devkit-5904422337438.html

with Waveshare CAN controller:

https://botland.com.pl/magistrala-can/6960-modul-z-interfejsem-can-sn65hvd230-waveshare-3945-5904422300715.html

As I mentioned - I tried to connect TX to G4 and RX to G5 (from previous setup with MQTT), and also according to new guide TX to G25 and RX to G21 - none work correctly. As far as I can tell - wiring was all right because it worked fine with MQTT version.

I actually just ordered another CAN module to check if the problem might be there, and also AZDelivery D1 Mini NodeMCU ESP32 to be as much compliant with the guide as possible. I will check it out on Wednesday and let you know the result.

vekexasia, 2023-05-22:
CAn you have a try right now?

vekexasia, 2023-05-22:
Apparently there was some recent changes in esp32_can which apparently introduced some issues. I needed to rollback to a prev commit.

Possibly related to https://github.com/collin80/esp32_can/issues/47

closed 2023-05-22 by vekexasia

SkullboyNL, 2023-05-22:
It is working now thankyou.

TroLoos, 2023-05-23:
What a great support, thank you! Awesome project @vekexasia.

commit 2fa1eb43 (2023-05-22): fix #47 rollback due to changes in esp32_can lib.