vekexasia/comfoair-esp32 #48: Installation Instructions for ESPHOME

marivo, 2023-05-10:
Dear creators of this cool project, I have been following it since December 2022 and understand now how to integrate my ESP32 board with HA via MQTT. That was great. However, now it is available also via ESPHOME project. Although I am new to ESPHOME I kinda understand that this the comfoair is a kind of plugin/custom device to the ESPHOME, similar as this is done also with EspEasy which I know very well. Nevertheless, no save me and others a little time, would it be possible to provide a brief list of instructions how to make the comfoair part of the ESPHOME to integrate with the HA? Thanks a lot in advance!

marivo, 2023-05-11:
I would be grateful for at least an indication how to involve the __init__.py file and create the GUI in the HA...

vekexasia, 2023-05-11:
You don't have to install anything from this repo. Nor download anything. You need to declare something in esphome yaml config file... 

check https://github.com/vekexasia/comfoair-esp32#esphome-conf

marivo, 2023-05-12:
So, meanwhile I read about ESPHOME project and installed the comfoair project on my ESP32 board. I could flash the project to the board from the command line and once compiled there, also from the VS Code. So far so gut. However, I noticed that the board is not running properly, The connection to the WiFi network is quite sporadic and the HomeAssistant does not recognise the flashed ESPHOME comfoair node by its autodiscovery feature. I have not connected he board to the Zehnder via the CAN sofar. Can this be the problem? Any comment on this issue would be helpful for me, thanks!!

lskuba, 2023-05-19:
Maybe your yaml file is incorrect, remember about "api:" and "ota:"  lines.

Please past your yaml file so we can check

marivo, 2023-05-21:
Thanks for the hint! Meanwhile I have progressed with ESPHOME, maybe even
too much, seen my little experience with it. Namely, I took arbitrary board
(not the mini with pins 21 and 25 made available to the outside for CAN
interfacing) and connected an OLED display to it, with the modification of
the yaml file to output up time on the OLED I also named the board
unconventionally, all that is apparently not liked by the Andrea's code. So
I am waiting for the mini board and will not do any modification to the
code as I intended originally. Just initialize the mini from the HA and put
the reference to the external component of Andrea in its yaml file. That
should work, finally:)

Besides: I have the impression that the CAN code of the Zehnder component
code is thread-blocking and if there is no connection to the ventilation
box via the CAN, it simply blocks the ESPHOME single execution thread with
all the consequences, e.g WiFi being disconnected. Can that be, can
somebody please confirm this?

Thanks!