vekexasia/comfoair-esp32 #49: Instructions about custom pcb

dan-s-github, 2023-05-28:
Thanks for making this project available, I ordered the KNX Interface and discovered that a lot of information that is available on the LAN Interface is missing. So this would be a perfect addition to my current system

I like the custom pub design and think this could also be used for other can based integrations.

Went to https://jlcpcb.com to order a set of fully assembled boards.

To order the fully assembled board I first uploaded the Gerber ZIP file and after activating and confirming PCB Assembly
a popup opens which requires to file to be uploaded

- BOM File
- CPL File

The BOM file is currently present in the repo but the CPL file is missing, is this also available or is there another way to get a fully assembled board?

dan-s-github, 2023-06-06:
ordered one of those to see if it would also work

https://shop.m5stack.com/products/atom-canbus-kit-ca-is3050g

vekexasia, 2023-06-06:
wow.. didnt see it. it would have been nice to have it would've spared somuch time :D

dan-s-github, 2023-06-06:
just missing the 12V to 5V converter which you added

vekexasia, 2023-06-06:
Ah. thne it's not good for me :D

dan-s-github, 2023-06-06:
can you add the full schematics of your custom pub to the repo for completeness?

vekexasia, 2023-06-06:
[image]

JoooostB, 2023-10-09:
@dan-s-github Did you manage to get a CPL file? Or have you walked the manual route?

closed 2024-03-11 by vekexasia