vekexasia/comfoair-esp32 #50: Missing Controls entities in ESPHome integration

lskuba, 2023-05-30:
I have followed instructions from README and after fixing problem with boot loop I was I able to read sensor entities, thank so much for that.

But I am missing controls entities to e.g. change temperature or fan speed.

I guest this is because of missing ClimateTraits clime controller, so it should be added into C code, or I am  not understanding something?

SkullboyNL, 2023-05-31:
You can use the service called: comfoair_send_command

```
The component exposes 2 services.

comfoair_send_command
comfoair_send_hex
NOTE: the prefix could differ depending on your esphome device name.

comfoair_send_command: allows one "command" parameter. The value of the param can be one of the following pre-bundled commands:

ventilation_level_0
ventilation_level_1
ventilation_level_2
ventilation_level_3
boost_10_min
boost_20_min
boost_30_min
boost_60_min
boost_end
auto
manual
bypass_activate_1h
bypass_deactivate_1h
bypass_auto
ventilation_supply_only
ventilation_supply_only_reset
ventilation_extract_only
ventilation_extract_only_reset
temp_profile_normal
temp_profile_cool
temp_profile_warm
```

marivo, 2023-06-04:
Hello all! Finally, I had time to to get this nice project up and running on my D1 ESP32 mini board, The HA now receives the data from the my Q350 and it has also discovered the two services that I can access under the Developer Tools tab to send commands. However, I am not so experienced with HA to immediately grasp how to set up this beautiful Dashboard that is shown on the main page of the project. Could someone please (!) provide some direction how to get this Dashboard done? Thanks a lot!

marivo, 2023-06-05:
Well, now I got behind the Dashboards in HomeAssitant and see how to create them. It would however take a lot of time to create something similar that Andrea already did. @Andrea:Would it be possible to share the HomeAssistant Dashboard code here such that we can adapt and use it without too much of hassle? That would be great!

JBN164, 2023-06-05:
Thanks a lot @vekexasia for your work, it works perfect. But...
I used the entity card to call a service, e.g. send the command to set the fan speed.
This way I need one card per fan speed. Smarter it would be to define the command in a dropdown and send this value with the call service option. Has anyone an idea to build something like that.

marivo, 2023-06-05:
> Thanks a lot @vekexasia for your work, it works perfect. But... I used the entity card to call a service, e.g. send the command to set the fan speed. This way I need one card per fan speed. Smarter [...]

This is what also I mentioned in the last post of mine above... In other words, as up to now, anyone who wants to fully integrate the Q350 ventilation system into HA using the Andrea's code needs to build his/her own Dashboard from the scratch. This can be done, of course, there are plenty of GUIs, including the dropdown menues etc, pp. But that would cost time that I personally unfortunately do not have right now... 

It would be therefore just great if someone who already created a control Dashboard for this project in HA would just share the contents of the corresponding HA yaml file... Anyone, please?

nebman, 2023-06-05:
I used this code in esphome to define some buttons and a very basic dropdown select field, but I'm sure there is a better way without using lambdas and an id. They will automatically show up in HA as new entities.

```
comfoair:
  id: comfo

button:
  - platform: template
    name: "Comfoair Boost 10m"
    id: comfoair_boost_10min
    on_press:
      then:
        - lambda: |- 
            id(comfo).send_command("boost_10_min");

select:
  - platform: template
    name: "Comfoair Stufe"
    id: comfoair_level
    optimistic: true
    options:
      - "Abwesend"
      - "Stufe 1"
      - "Stufe 2"
      - "Stufe 3"
    set_action:
      - lambda: |- 
          if (x == "Stufe 1") {
            id(comfo).send_command("ventilation_level_1");
          }
          if (x == "Stufe 2") {
            id(comfo).send_command("ventilation_level_2");
          }
          if (x == "Stufe 3") {
            id(comfo).send_command("ventilation_level_3");
          }
          if (x == "Abwesend") {
            id(comfo).send_command("ventilation_level_0");
          }
```

marivo, 2023-06-19:
> > I used this code in esphome to define some buttons and a very basic dropdown select field, but I'm sure there is a better way without using lambdas and an id. They will automatically show up in HA as new entities.
> 
> thanks @nebman it works very good.
> 
> Based on [https://community.home-assistant.io/t/zehnder-comfoconnect-component/82522/67](url) I build now a card for the integration of my Comfoair Q350 -->
> 
> [image]
> This is my Code:
> 
> ```
> type: picture-elements
> image: /local/Bilder/comfo4.png
> elements:
>   - type: state-label
>     entity: sensor.zehnder_lueftung_supply_fan_flow
>     style:
>       top: 33%
>       left: 73%
>       color: grey
>   - type: state-label
>     entity: sensor.zehnder_lueftung_exhaust_fan_flow
>     style:
>       top: 33%
>       left: 27%
>       color: grey
>   - type: state-label
>     entity: sensor.zehnder_lueftung_extract_air_temp
>     style:
>       top: 12%
>       left: 88%
>       color: black
>   - type: state-label
>     entity: sensor.zehnder_lueftung_extract_air_humidity
>     style:
>       top: 18%
>       left: 88%
>       color: black
>   - type: state-label
>     entity: sensor.zehnder_lueftung_post_heater_temp_after
>     style:
>       top: 40%
>       left: 88%
>       color: black
>   - type: state-label
>     entity: sensor.zehnder_lueftung_supply_air_humidity
>     style:
>       top: 46%
>       left: 88%
>       color: black
>   - type: state-label
>     entity: sensor.zehnder_lueftung_outdoor_air_temp
>     style:
>       top: 12%
>       left: 12%
>       color: black
>   - type: state-label
>     entity: sensor.zehnder_lueftung_outdoor_air_humidity
>     style:
>       top: 18%
>       left: 12%
>       color: black
>   - type: state-label
>     entity: sensor.zehnder_lueftung_exhaust_air_temp
>     style:
>       top: 40%
>       left: 12%
>       color: black
>   - type: state-label
>     entity: sensor.zehnder_lueftung_exhaust_air_humidity
>     style:
[... 205 lines cut ...]
>         style:
>           top: 83%
>           left: 70%
>           color: lightblue
>           '--mdc-icon-size': 35px
>         tap_action:
>           action: call-service
>           service: esphome.zehnder_lueftung_send_command
>           target: {}
>           data:
>             command: auto
>       - type: state-label
>         entity: sensor.zehnder_lueftung_operating_mode
>         icon: mdi:thermometer-low
>         style:
>           top: 93%
>           left: 70%
>           color: lightblue
> ```
> 
> The only Problem with this is, that there are two manual operating modes (unlimited and limited). The unlimited is set after send the command "manual", but if you set e.g. the fan speed in the unlimited manual mode it will be set to the limited mode which changes to auto after a couple of hour. I tried to fix it with the call_command "manual" again in the limited mode, but it doesn't change to unlimited, it's still limited.
> 
> Another little Problem is that I don't really know which entity is for the in-house temperature. It should be the supply_air_temp but this doesn't exist. The post_heater_temp_after is I think the nearest Temp. for the inside. My next step will be to add some sensors in our house and build a avarage entity for this

Today I had time to play around with your Lovelace code and noticed that  binary_sensor.lueftung_stufe_0,  binary_sensor.lueftung_stufe_1,  binary_sensor.lueftung_stufe_2 and  binary_sensor.lueftung_stufe_3 need further definitions. Would it be possible to post the definitions here? Thanks!

JBN164, 2023-06-20:
> Today I had time to play around with your Lovelace code and noticed that binary_sensor.lueftung_stufe_0, binary_sensor.lueftung_stufe_1, binary_sensor.lueftung_stufe_2 and binary_sensor.lueftung_stufe_3 need further definitions. Would it be possible to post the definitions here? Thanks!

I defined the binary sensors for changing color of the fan icon to show the current level.
I wrote this in the configuration.yaml:

````
template:
    - binary_sensor:
        - name: "Lueftung_Stufe_0"
          unique_id: "Lueftung_Stufe_0"
          state: "{{ is_state('sensor.zehnder_lueftung_fan_speed','0') }}"
          
        - name: "Lueftung_Stufe_1"
          unique_id: "Lueftung_Stufe_1"
          state: "{{ is_state('sensor.zehnder_lueftung_fan_speed','1') }}"
          
        - name: "Lueftung_Stufe_2"
          unique_id: "Lueftung_Stufe_2"
          state: "{{ is_state('sensor.zehnder_lueftung_fan_speed','2') }}"
          
        - name: "Lueftung_Stufe_3"
          unique_id: "Lueftung_Stufe_3"
          state: "{{ is_state('sensor.zehnder_lueftung_fan_speed','3') }}"

Peeka49, 2023-07-11:
Really nice card I tried your code but I'm missing some info on the card, Bypass info, filter info, remote info, are unknown. Do you know why I'm missing those info?

Thanks for all the work.

marivo, 2023-07-11:
> Really nice card I tried your code but I'm missing some info on the card, Bypass info, filter info, remote info, are unknown. Also when trying to change fan speed, I have service call fail.
> 
> Do you know why I'm missing those info and why the commands are not sent correctly?
> 
> Thanks for all the work.
> 
> [image]

At least in my case, this information appears with time passing after restart. However, you may need switch off and on the Q350 so that the software can properly initialize the variables that are missing in your example. The lack of information in your example is not the problem of the software, imho.

Peeka49, 2023-07-13:
Thanks Marivo, all info are now displayed after rebooting the Q350. Remaining problem is when trying to select fan speed, I have this error message: "failed to call service esphome/comfoair_comfoair_select. connection lost"

Any idea on why it is not working?

Thanks for your help.

marivo, 2023-07-13:
> Thanks Marivo, all info are now displayed after rebooting the Q350. Remaining problem is when trying to select fan speed, I have this error message: "failed to call service esphome/comfoair_comfoair_select. connection lost"
> 
> Any idea on why it is not working?
> 
> Thanks for your help.

That is strange. I have not tried manual fan selection speed yet, since switching auto-manual works fine for me and makes me happy. Maybe Andrea can provide a hint for the error you have?

vekexasia, 2024-03-11:
Hello all sorry for late reply.- Seems like you encountered some kind of integration bug between HA and esphome component?

I regularly use the fan speed control in my dashboard with no problem

marivo, 2024-03-12:
> Hello all sorry for late reply.- Seems like you encountered some kind of integration bug between HA and esphome component?
> 
> I regularly use the fan speed control in my dashboard with no problem

Dear Andrea, thanks a lot anyway. Could you please tell me what is the best way to install your plugin in the HA? Via the HACS? The thing is that if I just change the esphome config file I see the component in the HomeAssistant but no GUI that you display on the main page of the GitHub.

vekexasia, 2024-03-13:
There is no UI provided with this project. But there is a doc folder with
some sample yaml you can check.

marivo, 2024-03-16:
I get it. Could not try it yet, but may be a quick question before I get to it... Will I get the same appearance as depicted on your github main page (screenshot below) ?

[image]

vekexasia, 2024-03-16:
Yeah more or less.

marivo, 2024-03-20:
Hi Andrea, just compiled the new firmware according to the example comfoair.yaml file from the repository. I cannot see any of the appearance as above on the HomeAssistant main webpage. I think it is because the screenshot that I have put above is actually a manually created  Lovelace graphical interface. Is that true? If yes, would it be possible to share the Lovelace code here so that we can reproduce exactly the same appearance (which I like a lot :))?

marivo, 2024-03-25:
Please...