vekexasia/comfoair-esp32 #52: auto mode not available

J1Buskens, 2023-06-16:
Hi I just upgraded to the ESPHOME version.
When I send the command 'auto' the unit doesn't seem to respond.
I am able to set other settings like e.g. : 'ventilation_level_3'.
I have used the old version with mqtt communication and there the auto command was working.

KrisDeWilde, 2023-11-13:
I have the same issue. Send command 'auto' doesn't respond. Everyting else works fine.

vekexasia, 2024-03-11:
which comfo unit do you have?

J1Buskens, 2024-03-11:
> which comfo unit do you have?

Comfoair Q350

vekexasia, 2024-03-11:
Can you try to set ventilation speed manually to 0. Wait for a bit till reported fan rpm in your dashboard is visibly lower and then switch to auto?

Does it modify the fan speeds?

KrisDeWilde, 2024-03-21:
Hello, my device is a Comfoair Q450.

I tried your suggestion (manual to 0 then to automatic) and it works.

I have to say that my original hardware configuration wasn't the best. I used a Velleman kit to reduce the unit's voltage to 3.3 V. After a few weeks it started missing a lot of values ​​and the unit disabled the CAN bus. After a few resets I started debugging my hardware and I discovered that the gnd of the device and the ground of the ESP were not quite the same, I saw a diode between them.
Later I fitted a separate USB power supply and connected the device ground to the ESP ground. Now it has been working like a charm for almost 3 months.

Thanks again for this great work!!

Maybe an idea for other users:
I have set my unit to always respond to moisture, even if it is not set to automatic. This causes the device to run at a higher speed when someone takes a bath.

vekexasia, 2024-07-02:
I am interested in the moisture trick. Are u using an external sensor or data from internal humidity as  input?

KrisDeWilde, 2024-07-02:
It is just a setting in the unit. You can set the response for humidity to: 
-"not"
-"only on auto"
-"always" 
I´m not sure all unit have this sensor, but mine does. When it responds to humidity, the air-flow increases.
Depending on what current ventilation level: 
In my case:
- level 1 goes from 165m³ to 255m³
- level 2 goes from 320m³ to 400m³
- level 3 is already max, so no change here.
I suggest level 2 as starting point, this responds very quickly. Level 1 is very slow (more the 10 min delay)
A made a link with the bathroom light, the unit is on level 2 when the light is on.

vekexasia, 2024-07-02:
Very interesting. which unit do u have ?

KrisDeWilde, 2024-07-04:
Zehnder Comfoair Q450

Tigerstrength, 2024-10-28:
First let me thank you @vekexasia! I use this integration to control my Q450 with great success. Only thing I miss is the option to set iT to auto mode and to set the fans to any percent (0-100%) bases on the actual needs. 

I think the 0-100% fans are complicated to achieve as that requires emulating the option box, but is there any change we can ‘repair’ the fact that auto mode is not working? I read that it was working in the mqtt version? Any hint about what to send to the unit to get it on automode?

jvanoverveldt, 2024-11-09:
I use this esphome setup with my Q450. When sending out the "auto" command the log gives this output:

`[09:04:06][W][component:237]: Component api took a long time for an operation (1002 ms).` 
`[09:04:06][W][component:238]: Components should block for at most 30 ms.` 

The unit is never set to "auto" even after setting ventilation_level_0 and waiting for some time.

When sending "ventilation_level_x"

``[09:13:51][W][component:237]: Component api took a long time for an operation (1004 ms).
[09:13:51][W][component:238]: Components should block for at most 30 ms.
[09:13:51][D][sensor:094]: 'fan_speed': Sending state 0.00000  with 0 decimals of accuracy
[09:13:51][D][comfoair.component:244]: ibua fan_speed 0.000000
[09:13:51][D][climate:396]: '' - Sending state:
[09:13:51][D][climate:399]:   Mode: HEAT
[09:13:51][D][climate:404]:   Fan Mode: OFF
[09:13:51][D][climate:419]:   Current Temperature: 18.90°C
[09:13:51][D][climate:425]:   Target Temperature: 23.00°C
[09:13:51][D][climate:428]:   Current Humidity: 55%
[09:13:51][W][component:237]: Component comfoair took a long time for an operation (91 ms).
[09:13:51][W][component:238]: Components should block for at most 30 ms.
``
And the state is set correctly

@vekexasia What is the reason it could work when setting it to zero first?

vekexasia, 2024-11-09:
@jvanoverveldt so setting vent level to zero and then sending auto works?

jvanoverveldt, 2024-11-09:
No, i was not able to use the workaround of setting the ventilation level to 0. That's the reason that i'm trying to understand the logic behind the trick.

pippolino79, 2025-01-31:
Hi, I think I discovered the problem with the AUTO mode:

the "comfoair_operating_mode" has 3 possible states:
- limited_manual
- unlimited_manual
- auto

When you are for example in state "auto" at "speed 1" and you decide to issue command "ventilation_level_2" comfoair swaps speed to level 2 but also swaps mode to "limited_manual".
I guess this is a mode in which after a while the state returns to "auto".

Now the issue/bug:
It is NOT POSSIBLE to go from "limited_manual" directly to "auto" by sending the "auto" command, I don't know why but it is so.
You need first go from "limited_manual" to "unlimited_manual" by issuing the "manual" and then you can go to "auto".

You can try the NON working sequence of commands when you're in auto:
- ventilation_level_2 (goes to state limited_manual)
- auto (remains in state limited_manual)

While this sequence works:
- ventilation_level_2 (goes to state limited_manual)
- manual (goes to state unlimited_manual)
- auto (goes to auto)

Clearly the second working sequence  is not intuitive, if I push AUTO it should go to AUTO and not hafo to go to manual before!!!

I hope this helps to fix the question...