vekexasia/comfoair-esp32 #53: ComfoAir 70

JBN164, 2023-06-19:
Does this project also work with a ComfoAir70. 

It doesn't support the LAN C Adapter but works with a RF + Connect box. 

Connection via Bus is available (A,B, + and -). 

[image]

[image]

vekexasia, 2023-06-19:
Unfortunately I can't answer that question. It looks like there is a can interface.

[image]

But if and on which extent this project would work it's unknown.

vekexasia, 2024-05-24:
Were u able to test it out?

vekexasia, 2024-07-02:
Hey @JBN164 any news?

JBN164, 2024-07-02:
@vekexasia 
No I didn't test it and have no further informations

vekexasia, 2024-07-02:
Ok, can I close the issue?

closed 2024-07-04 by vekexasia