vekexasia/comfoair-esp32 #55: Unknown entity state after restart.

SzymaneKwr, 2023-09-06:
Hello, I have a question about the entity state after ESP reset. All entities are in an unknown state until they are replaced or updated. Is there any way to make entities show their current state after a reset/update?
[image]

xhess, 2023-09-06:
Thank you for asking this question. I have got the same problem. I think I know what the problem is and that currently there is no solution in this code. As I understand that comfoair-esp32 just listens for messages on the can-bus. The ventilation unit only sends the values when they change. A quick "fix" would be to restart the ventilation unit or you wait until the value changes...

While writing this comment there is currently the possibility to send raw hex commands like described below.

But a real solution would be that the esp32 could ask the ventilation unit for some/all sensor values like described here:
https://github.com/michaelarnauts/comfoconnect/blob/master/PROTOCOL-RMI.md#general-commands

> This command reads a single property identified by the Unit, SubUnit and Property Id. Each writable property may have a definitionrange & recommended step size which are also accesible
> The syntax is: 01 UnitId SubUnitId GetType PropertyId The answer is depended on the type of the property and the GetType. If the value, range and step (everything) is requested, the answer is: Value, lower boundry of range, upper boundry of range, step size.

The ESP could know that there are unknown values and ask specific for these or it could ask for all values.

albertofralbe, 2024-06-17:
Hi, I was wondering if this issue has already been addressed by someone? If so, would you mind to share? Thanks

vekexasia, 2024-06-17:
Hey all,

as far as I know nobody addressed this. tbh i still struggle to understand the usecase of this. I've mine up and running for weeks and apart from esphome updates my esp32 never reboots... maybe some of you cuts power to their VMC unit?

vekexasia, 2024-07-02:
Can u guys let me know if this is working for u?

albertofralbe, 2024-07-02:
Where is the PDOID list?

vekexasia, 2024-07-02:
> Where is the PDOID list?

https://github.com/vekexasia/comfoair-esp32/blob/master/components/comfoair/__init__.py#L26

Also can u please take a look @ https://github.com/vekexasia/comfoair-esp32/issues/74 I am trying to build a compatibility list

commit 242346e5 (2024-07-02): added selfupdate and services to avoid unavailable

fix #55
closed 2024-07-04 by vekexasia