vekexasia/comfoair-esp32 #57: Sending commands not working with the esphome version

Paddy-van-CH, 2023-09-26:
Trying to send any command (fan speed, temp. profile etc.) reports following errors in the log: 

[W][component:204]: Component api took a long time for an operation (1.00 s).
[W][component:205]: Components should block for at most 20-30ms.

pat1300, 2023-10-05:
I got the same issue when I first used a wrong SN65HVD230 [https://nl.aliexpress.com/item/32686393467.html](url)
Then I changed for the Waveshare model [https://www.amazon.com.be/dp/B0B5DTN62K](url) and it is working. I don't understand if differences.
Also be sure to check if you use the correct pins of ESP for TX (25) and RX  (21). Attention: the old MQTT version use different pin numbers !

Paddy-van-CH, 2023-10-07:
Thanks @pat1300. I'll try to get my hands on a Waveshare model then.

WeaselDev, 2023-12-28:
Hi, i know, this is an old topic but i have the same issue. 
Got the small SN65HVD230 board with an VP230 AFC3 first. It is able to read values but not send any commands.
So i got a Waveshare SN65HVD230 today with an VP230 OASYV soldered on. 
This one is not even able to read data from the CAN but.
Maybe it is just defect. Can anyone confirm that the OASYV variant should be working or are there so many variants out there that none of us has twice the same chip?

closed 2024-05-24 by vekexasia