vekexasia/comfoair-esp32 #58: RX/TX pinout for node mcu ESP32s wroom

KASR, 2023-10-18:
Hi, 

I have an ESP32 from node mcu (see the figure below for the pinout and the board specifics.

I think the pinout is not exactly the same as what's mentioned in the readme. Is my understanding that the pin 21/25 mentioned in the readme are the TX0 and RX0 pins in the figure below?

[image]

In other words, would this be the correct connection:

```
---------------+        +---------------+                 +-------------+
(oran/red)12V  o--------o IN+      OUT+ o-----------------o VIN         |
               |        |   [LM2596S]   |                 |             |
(brown)   GND  o--------o IN-      OUT- o-----------------o GND         |
               |        +---------------+                 |             |
    [RJ45]     |                                          |   [ESP32]   |
  [ComfoAir]   |                                          |             |
               |        +-------------------------+       |             |
(w/blue) CAN_L o--------o CAN_L               3v3 o-------o 3v3         |
(blue)   CAN_H o--------o CAN_H               GND o-------o GND         |
---------------+        |                         |       |             |
                        |   [SN65HVD230]   CAN TX o-------o TX0         |
                        |                  CAN RX o-------o RX0         |
                        +-------------------------+       +-------------+
```

vekexasia, 2023-10-18:
Hey you can use whwatever pins you prefer. But definitely pin 21 is no tx0
:) there is pin21 in your pic.

-- 

Andrea Baccega
[image: ***@***.***
[image: ***@***.***
[image: Personal site] <http://www.andreabaccega.com/> [image: Linkedin
Profile] <http://it.linkedin.com/in/andreabaccega> [image: Facebook]
<https://www.facebook.com/andrea.baccega> [image: Google+]
<https://plus.google.com/109217393200753135791> [image: Twitter]
<http://twitter.com/veke87> [image: StackOverflow]
<http://stackoverflow.com/users/314407/andrea-baccega> [image: GitHub]
<https://github.com/vekexasia>

KASR, 2023-10-18:
ah ok i understand, so i do not need to use any RX/TX pin's from the ESP32 for the can communication

to use the code from the repository directly i need to connect the SN65HVD230 to the P21 and P25 on the board (circled in red below)

[image]

vekexasia, 2024-07-02:
Sorry for being late in reply. Yes u can use that as well. Did u succeed in the end? I would also like to know your setup in #74 :)