vekexasia/comfoair-esp32 #59: comfoair ESP32 - MQTT supported in ESPhome?

tudstudent, 2023-11-01:
Sorry, searching did not supply an answer.
I see there is a discrepancy in commit dates between the mqtt tree and the normal tree.
Furthermore, normal tree is esphome enabled, cannot find if mqtt tree can be used the same?

Additionally, I was searching for mqtt config details in esphome, also no luck so far.

Would you mind clarifying these details (or point in the right direction)

fuatakgun, 2023-11-01:
esphome version does not require mqtt because it is directly connecting with home assistant using home assistant components.

vekexasia, 2023-11-01:
> Yes, I know, but for certain integration tasks it might be helpful.
> So just looking if supported.

Hey you can hook mqtt through home assistant if needed pushing data as soon
as ha receives it

tudstudent, 2023-11-01:
Correct, have that already.
But it depends on HA.
Ideally would be to have it pushed to MQTT and then picked up by whatever UI/Tool I want.
(If HA is down for whatever reason the stable MQTT remains online and therefore data collection unaffected)

vekexasia, 2023-12-22:
@tudstudent sorry for late response. you can use mqtt component and publish everything there if you want.

closed 2024-03-11 by vekexasia