vekexasia/comfoair-esp32 #63: Instruct ComfoAir to resend all data on the CAN bus

sodre90, 2023-12-22:
Thanks for your great work! Is there a HEX command that can be used to force the ComfoAir to send all data, including bypass state and percentage? When my Home Assistant restarts, the data appears blank, and it would be great to send a HEX command to the ComfoAir system to publish the data, just as it does during it's start-up. I tried to send hex commands with read flag, but no luck.

vekexasia, 2023-12-22:
hey @sodre90 if there's any I am not aware of it unfortunately...

WeaselDev, 2023-12-23:
I have the original Zender ComfoConnect LAN C interface running and integrated into Home Assistant. With this interface, all the data is available after Home Assistant restarts. Maybe there is something to find in the Home Assistant Zehner ComfoAirQ integration. 
Btw. great project :-)

closed 2024-05-24 by vekexasia

vekexasia, 2024-07-02:
hey this was fixed please check #55