vekexasia/comfoair-esp32 #66: Error in communication after connecting ESP to ComfoAir (wont even start)

K3iVi, 2024-02-05:
I used standard ESP, but dc stepdown is LM2596. Other than that is everything by manual. ESP is recognized in HA. But when I connect it to Comfoair there is an error of communication. When I try to restart it do not even boot, I have to disconnect 12V from Comfo to stepdown module in order to comfoair to start normally. 
Double check with multimeter and stepdown module is on 3,3V output voltage. 
I am also using comfosense-c55 module, but when I disconnect that it still not working only with ESP.

Can be some overcurrent on comfoair or something? Or I missed something?
I will try external power supply for esp.

[image]

vekexasia, 2024-02-05:
I doubt it might be an overcurrent protection tbh. if esp32 does not boot then it might be 3.3 is not stable and it goes in brownout detection.

A short could also explain it but i can't see it from this pic. U sure your board has 3v3 on that vcc pin? I see a voltage regulator on that module so maybe that is +5v?

K3iVi, 2024-02-05:
ComfoAir unit won't start up after restart when EPS is connected to the ComfoAir board, thas is the problem. I have to unplug 12V from the board to get it working again. I triple checked wiring... Stepdown is set to 3,3V checked with multimeter. Connected to ESP to 3.3V.
Should I increase voltage? Or any other guesses?

nebman, 2024-02-05:
Double check the ESP side CAN_RX and CAN_TX pins, I had the same behaviour when the default pins changed, probably blocks the bus otherwise

K3iVi, 2024-02-07:
Thank you :) this was it. I had the connection right but by a different/wrong scheme where it was connected to D4/D5. But it should be to 21 and 25. Reconnected and works.

vekexasia, 2024-03-11:
Glad to hear it worked @K3iVi closing this issue