vekexasia/comfoair-esp32 #67: Deprecated warning for esp_intr.h

wauswaus, 2024-03-06:
During compile in ESPHome version 2024.2.2:
`Compiling .pioenvs/esp-growatt-modbus/lib2a8/esp32_can/esp32_can_builtin_lowlevel.cpp.o
In file included from .piolibdeps/esp-growatt-modbus/esp32_can/src/esp32_can_builtin_lowlevel.cpp:32:
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_hw_support/include/esp_intr.h:8:2: warning: #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead [-Wcpp]
 #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead`

vekexasia, 2024-03-06:
hey, thanks for reporting. do you feel up for a pullrequest?

-- 

Andrea Baccega
[image: ***@***.***
[image: ***@***.***
[image: Personal site] <http://www.andreabaccega.com/> [image: Linkedin
Profile] <http://it.linkedin.com/in/andreabaccega> [image: Facebook]
<https://www.facebook.com/andrea.baccega> [image: Google+]
<https://plus.google.com/109217393200753135791> [image: Twitter]
<http://twitter.com/veke87> [image: StackOverflow]
<http://stackoverflow.com/users/314407/andrea-baccega> [image: GitHub]
<https://github.com/vekexasia>

wauswaus, 2024-03-06:
mmm looks like its not in your code but somewhere in the libs. Sorry for the trouble...

closed 2024-03-06 by wauswaus