vekexasia/comfoair-esp32 #68: Error on comfoconnect

guillaum111, 2024-03-09:
Hi, 

Just finish the wiring my ESP32 to a Zehnder Comfoair Q600 premium. The communication seems to work as i receive some value. But the comfoair show '' error on the comfoconnect network, check the wiring". As soon as the message appear, I receive updated value after a long interval (+- every 1 minute ) 

Did someone already encounter this issue ? 

Thanks

guillaum111, 2024-03-29:
Sorry for the late answer, I had some health issue. I have nothing else connected to bus. When the error is not detected, it seems to work has i was able to send a command. I use another Can transceiver that i had in stock. maybe that's the issue. (https://www.amazon.com.be/dp/B0B1WW5XHV?psc=1&ref=ppx_yo2ov_dt_b_product_details). I will order the write transceiver.

guillaum111, 2024-04-03:
Problem solved, 2 of the 3 can transceiver i add are defect. With the good one it's working fine. I still replaced it by a wave shark to avoide issue

vekexasia, 2024-05-24:
Perfect let me close the issue :)

closed 2024-05-24 by vekexasia