vekexasia/comfoair-esp32 #72: Send only selected items

Schildkroetenprogrammierer, 2024-04-02:
My MQTT broker gets a lot of unnecessary information: each second "next_fan_change" and "exhaust_fan_flow", "exhaust_fan_speed", "supply_fan_flow", "supply_fan_speed", "power_consumption_current" and "avoided_heating_actual". 

As the changes are only minimal (+/- 1 m³/h, 5 RPM,...) they are not important in each second. One update per  minute or per 5 minutes would be enough to reduce traffic on the WLAN and the MQTT broker. 

How can I get this behaviour? Is there a setting in ESPHome or in Comfoair-ESP32 possible?

vekexasia, 2024-04-02:
I guess you can use filters for that? https://esphome.io/components/sensor/index.html

vekexasia, 2024-05-24:
closing as no more activity :)

closed 2024-05-24 by vekexasia