vekexasia/comfoair-esp32 #74: Showcase Ur setup and compatibility list

vekexasia, 2024-05-24:
Hello, it would be nice if people share their setup and their comfo machine model number so I can keep track of the project compatibility list.

hb0nes, 2024-10-28:
I made this work with my Q350, with a LilyGo TCAN-485.
I am a complete newb, but somehow managed anyway after some trial & error with a board that wasn't listed on this repo.
I thought it might help others.

I:
- wired the GND (black) from the Q350 to both CAN_GND and - for the board.
- wired 12V (red) to +.
- wired Yellow to CAN_H and white to CAN_L.

Added this to ESPHome yaml:

```
# For me, setting these to off made everything work.    
switch:
  - platform: gpio
    id: can_se
    pin: 
      number: 23
      inverted: false
    name: "can_se"
  - platform: gpio
    id: can_en
    pin: 
      number: 16
      inverted: false
    name: "can_en"

external_components:
  - source: github://vekexasia/comfoair-esp32
    components: [ comfoair ]

comfoair:
  id: comfo
  rx_pin: GPIO26
  tx_pin: GPIO27

# Enable Web server
web_server:
  port: 80

button:
  - platform: template
    name: "Boost 10m"
    id: comfoair_boost_10min
    on_press:
      then:
        - lambda: |- 
            id(comfo).send_command("boost_10_min");
  - platform: template
    name: "Boost 60m"
    id: comfoair_boost_60min
    on_press:
      then:
        - lambda: |- 
            id(comfo).send_command("boost_60_min");
  - platform: template
    name: "Boost End"
    id: comfoair_boost_end
    on_press:
      then:
        - lambda: |- 
            id(comfo).send_command("boost_end");

select:
  - platform: template
    name: "Ventilation Level"
    id: comfoair_ventilation_level
    optimistic: true
    options:
      - "Disabled"
      - "Level 1"
      - "Level 2"
      - "Level 3"
    set_action:
      - lambda: |- 
          if (x == "Level 1") {
            id(comfo).send_command("ventilation_level_1");
          }
          if (x == "Level 2") {
            id(comfo).send_command("ventilation_level_2");
          }
          if (x == "Level 3") {
            id(comfo).send_command("ventilation_level_3");
          }
          if (x == "Disabled") {
            id(comfo).send_command("ventilation_level_0");
          }
  - platform: template
    name: "Mode"
    id: comfoair_mode
    optimistic: true
    options:
      - "auto"
      - "manual"
    set_action:
      - lambda: |- 
          if (x == "auto") {
            id(comfo).send_command("auto");
          }
          if (x == "manual") {
            id(comfo).send_command("manual");
          }
```

jvanoverveldt, 2024-11-09:
I can confirm a working setup with a Q450:

I used these components:
-AZ-Delivery LM2596S Buck DC-DC
-Waveshare SN65HVD230 Can
-AZDelivery NodeMCU ESP32 S (i used this one because it has an external antenna connection)
-external antenna

The configuration is not able to set "auto" mode. I added my insights in #52 

My current yaml is:

´´
external_components:
  - source: github://vekexasia/comfoair-esp32
    components: [ comfoair ]

esphome:
  name: zehnder
  friendly_name: zehnder
  min_version: 2024.6.0
  name_add_mac_suffix: false

esp32:
  board: nodemcu-32s
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

wifi:
  # Set up a wifi access point
  manual_ip:
    static_ip: 192.168.0.60
    gateway: 192.168.0.1
    subnet: 255.255.255.0
  ssid: !secret wifi_ssid
  password: !secret wifi_password

time:
  - platform: homeassistant
    id: homeassistant_time

# Enable Web server
web_server:
  port: 80

comfoair:
    id: comfo
    rx_pin: GPIO3
    tx_pin: GPIO1

gytisgreitai, 2025-02-07:
https://www.aliexpress.com/item/1005006734943258.html for me doesn't work with this one SN65HVD230. 350Q does not even boot up, and after unplugging gets into safe mode and asks how to boot 

Works fine with old usb-to-can adatper though

TriggrHappy, 2025-02-07:
I can confirm this. Only original waveshare worked for me.

> https://www.aliexpress.com/item/1005006734943258.html for me doesn't work with this one SN65HVD230. 350Q does not even boot up, and after unplugging gets into safe mode and asks how to boot
> 
> Works fine with old usb-to-can adatper though

gytisgreitai, 2025-02-07:
Where does one buy Waveshare? Directly from them ?

jvanoverveldt, 2025-02-07:
I bought it from Amazon.

fallenartist, 2026-03-14:
@zweefan wow can you elaborate on the design? Is this a custom PCB with a soldered ESP board?