vekexasia/comfoair-esp32 #76: compiling error

mattapo, 2024-07-05:
Hi, i use  Esphome version 2024.6.6,

this error occurs during compilation:

"
.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bitset: In member function 'void std::bitset<_Nb>::_M_copy_from_ptr(const _CharT*, std::size_t, std::size_t, std::size_t, _CharT, _CharT)':
src/esphome\components\comfoair\comfoair.h:14:18: error: expected unqualified-id before '(' token
 #define min(a,b) ((a) < (b) ? (a): (b))
                  ^
src/esphome\components\comfoair\comfoair.h:14:18: error: expected unqualified-id before '(' token
 #define min(a,b) ((a) < (b) ? (a): (b))
                  ^
src/esphome\components\comfoair\comfoair.h:14:18: error: expected unqualified-id before '(' token
 #define min(a,b) ((a) < (b) ? (a): (b))
"

Can you help me?

vekexasia, 2024-07-05:
hey @mattapo let me give it a try and get back to u also u can join [discord](https://discord.gg/dUSSRezJXC) if you prefer a more instant response. most likely it's a bug I introduced previous days

vekexasia, 2024-07-05:
can u show me your configuration?

mattapo, 2024-07-05:
Any simple configuration, but also your example yaml.

vekexasia, 2024-07-05:
If u can share your configuration (full) it would be better. Also first try with just comfoair:

vekexasia, 2024-07-06:
@mattapo which version of esphome are u using?

mattapo, 2024-07-06:
I'm using version 2024.6.6

vekexasia, 2024-07-08:
can you try to:

1 have this minimal configuration:
```
comfoair:
  global_filters:
    - throttle: 10
```

NOTE: if you need to set **rx and tx** pins do that 

2. clean build files using either `esphome clean file.yaml` or by using the three dots in the interface like shown here:

[image]

and after that the error should go away as we found in discord with another user

closed 2024-07-08 by mattapo