vekexasia/comfoair-esp32 #77: Showing up in HA but entity stay in UNKNOWN state

bcornelis, 2024-07-05:
Dear,

I have the following setup:
- Zehnder ComfoAir Q 600 (not listed as compatible in this repo afaik, but expecting it to work)
- Sn65hvd230 Vp230 Can reader: https://nl.aliexpress.com/item/1005006438437964.html
- esp32cam: 
[image]

Following connections are made:
- can RX --> GPIO3
- can TX --> GPIO1
- can H --> blue wire
- can L --> white/blue wire

The esp32 is properly setup in esphome showing up and I see al the entities, but their state is unknown:
[image]

So their must be something not properly setup. What's the best way to start debugging? Can I enable CAN logging to make sure the connection to the Zehnder device is working?

Thanks!

bcornelis, 2024-07-05:
As extra information, I took the comfo config from following file: https://github.com/vekexasia/comfoair-esp32/blob/master/docs/esphome_sample.yaml, only changing the pin numbers

vekexasia, 2024-07-05:
Can u check the logs and post them here?

bcornelis, 2024-07-05:
I restarted the esp32, and then see those logs appear:

`[16:48:01][C][logger:185]: Logger:
[16:48:01][C][logger:186]:   Level: DEBUG
[16:48:01][C][logger:188]:   Log Baud Rate: 115200
[16:48:01][C][logger:189]:   Hardware UART: UART0
[16:48:01][C][mdns:115]: mDNS:
[16:48:01][C][mdns:116]:   Hostname: esphome-web-394a54
[16:48:01][C][esphome.ota:073]: Over-The-Air updates:
[16:48:01][C][esphome.ota:074]:   Address: esphome-web-394a54.local:3232
[16:48:01][C][esphome.ota:075]:   Version: 2
[16:48:01][C][safe_mode:018]: Safe Mode:
[16:48:01][C][safe_mode:020]:   Boot considered successful after 60 seconds
[16:48:01][C][safe_mode:021]:   Invoke after 10 boot attempts
[16:48:01][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[16:48:01][C][api:139]: API Server:
[16:48:01][C][api:140]:   Address: esphome-web-394a54.local:6053
[16:48:01][C][api:144]:   Using noise encryption: NO
[16:48:02][D][comfoair.component:166]: update_next 278 - iterator 31
[16:48:03][D][comfoair.component:166]: update_next 290 - iterator 32
[16:48:04][D][comfoair.component:166]: update_next 291 - iterator 33
[16:48:05][D][comfoair.component:166]: update_next 292 - iterator 34
[16:48:06][D][comfoair.component:166]: update_next 293 - iterator 35
[16:48:07][D][comfoair.component:166]: update_next 294 - iterator 36
[16:48:08][D][comfoair.component:166]: update_next 369 - iterator 37
[16:48:09][D][comfoair.component:166]: update_next 370 - iterator 38
[16:48:10][D][comfoair.component:166]: update_next 371 - iterator 39
[16:48:11][D][comfoair.component:166]: update_next 372 - iterator 40`

vekexasia, 2024-07-05:
This looks like u're not reallly receiving anything on canbus

bcornelis, 2024-07-05:
There are two assumptions:
- I haven't connecties the 12v line as I'm powering the ESP32 through An external 5V power supply
- I haven't enabled anything on the Zehnder tot 'activate' the bus
Are those assumptions correct?

bcornelis, 2024-07-05:
I've solved the issue by not using the RJ45 connector, but using the canbus wires directly. I do get the expected results now: 
[image]

Thanks for your help!

closed 2024-07-05 by bcornelis

vekexasia, 2024-07-05:
Glad to hear @bcornelis can u also comment on #74 ?