vekexasia/comfoair-esp32 #78: compilation error

saker88, 2024-07-07:
Im getting error and I do not know how to fix this ?

[away_indicator] is an invalid option for [comfoair]. Please check the indentation.

ive tried to rewrite that part , space check and nothing seems to help.. 

please help :D im blind and lost in fog here.

vekexasia, 2024-07-07:
Hey @saker88 can u share ur config please?

saker88, 2024-07-07:
esp config

`esphome:
  name: comfoair
  friendly_name: comfoair

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "6xoH9BDiFKQT+D9M4PfCufjaH0M7duaJOKlVMqiYlJM="

ota:
  - platform: esphome
    password: "71fcd5be07c55b0dfe28e97cf74d42bb"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Comfoair Fallback Hotspot"
    password: "GymQmucjlJg9"

captive_portal:

external_components:
  - source: github://vekexasia/comfoair-esp32
    components: [ comfoair ]

comfoair:
  id: comfo
  rx_pin: GPIO5
  tx_pin: GPIO4
  supply_fan_speed:
    name: supply_fan_speed
    disabled_by_default: false
    force_update: false
    unit_of_measurement: rpm
    accuracy_decimals: 0
    filters:
      - or:
        - throttle: 300s
        - delta: 100.0
  fan_speed:
    name: fan_speed
    disabled_by_default: false
    force_update: false
    unit_of_measurement: ''
    accuracy_decimals: 0
    filters:
      - throttle: 300s
  next_fan_change:
    name: next_fan_change
    disabled_by_default: false
    force_update: false
    unit_of_measurement: s
    accuracy_decimals: 0
    filters:
      - or:
        - throttle: 300s
        - delta: 180.0
[... 307 lines cut ...]
          }
          if (x == "Stufe 2") {
            id(comfo).send_command("ventilation_level_2");
          }
          if (x == "Stufe 3") {
            id(comfo).send_command("ventilation_level_3");
          }
          if (x == "Abwesend") {
            id(comfo).send_command("ventilation_level_0");
          }
  - platform: template
    name: "Mode"
    id: comfoair_mode
    optimistic: true
    options:
      - "auto"
      - "manual"
    set_action:
      - lambda: |- 
          if (x == "auto") {
            id(comfo).send_command("auto");
          }
          if (x == "manual") {
            id(comfo).send_command("manual");
          }`

vekexasia, 2024-07-08:
hey can u check this comment here ?  https://github.com/vekexasia/comfoair-esp32/issues/76#issuecomment-2213244441

you do not really need all the other configuration elements now you can specify a global filter like so:

```yaml
comfoair:
  global_filters:
    - throttle: 10s
```

mattapo, 2024-07-08:
> Im getting error and I do not know how to fix this ?
> 
> [away_indicator] is an invalid option for [comfoair]. Please check the indentation.
> 
> ive tried to rewrite that part , space check and nothing seems to help..
> 
> please help :D im blind and lost in fog here.

"away_indicator" is just "away". Edit the yaml file and the error will not appear

vekexasia, 2024-07-08:
hey @mattapo  i already removed all the entries from the sample most people dont need throttling per single sensor https://github.com/vekexasia/comfoair-esp32/blob/8d63d76fc53c933127928f9cec91cd4bae2cf187/docs/esphome_sample.yaml#L5C1-L8C16

saker88, 2024-07-10:
Hey, I managed to solve my issue by omitting the away_indicator and rewriting whole code once more. Now, I need to handle the wiring. I've noticed that you're using the M5Stack Atom (I assume) and in your code, you're referring to GPIO5 and GPIO4. On the PCB, you're connecting to pins G21 and G25. How do these correlate with the ESP32 WROOM dev board? Are they the same, or do I need to redefine them to different pins?

Thanks for updating the sample code and explaining the basics. I'm new to the HA system and YAML, and I'm trying to learn and understand by running shared projects. I appreciate your patience with us "noobs." It's great that even seemingly silly problems or questions don't go unanswered and are addressed in a professional manner.

vekexasia, 2024-07-10:
Hey yes. I am using the custom PCB with the m5stack atom. But the example code was made by another person. You can use the pins you want. If you can choose, checkout the datasheet for where parts are (skip uart0)

closed 2024-07-16 by saker88