vekexasia/comfoair-esp32 #87: fan_speed automatically changes every 2 hours

TroLoos, 2025-01-23:
Hello, I have Zehnder ComfoAir Q450 that I control with comfoair-esp32 but also via PWM -> 0-10v -> Option Box.

For this setup to work (leave control precisely to 0-10v input), I need to have ventilation level set to **off** and only then my steering is accurate. If I have any other level chosen on climate entity, my 0-10v input will be added on top of that (simplifying - level 1 is 30% + 3v = 60% cumulative level).

Why is that happening? I created a watchdog that changes is back to **off** whenever it is changed.

You can see the illustration of that below (easy to see the moment I enabled my watchdog):

[image]

vekexasia, 2025-01-23:
so it goes on lvl 2 every 2h?

TroLoos, 2025-01-23:
> so it goes on lvl 2 every 2h?

Yes, exactly every 2 hours... take a look:

[image]

When my watchdog changes fan speed back to 0, it will try to reach lvl 2 exactly after 2 hours.

To be fair - I am not sure what component tries that exactly, for sure it is not the 0-10v addon as the setting is changed on a unit-level. Maybe my Q450 does that on its own? I am not sure, on weekend I can try to unplug esp32 and see if it raises the speed again.

vekexasia, 2025-01-23:
yes please try that and let me know

TroLoos, 2025-01-25:
> yes please try that and let me know

Yup, it does it on its own... I have a service scheduled, will ask why is that happening, probably some auto setting is on that should be off. But it's great that I can use this integration to automate turning it off.

Thanks again, awesome project, one of the best integrations working @ my house.

closed 2025-01-25 by TroLoos

vekexasia, 2025-01-25:
Glad to hear that. Feel free to open a new issue in case you need