vekexasia/comfoair-esp32 #88: Set fan speed to zero

Schildkroetenprogrammierer, 2025-04-20:
How can I set the fan speed to 0 rpm? In case of bad air or a fire I would like to turn off the fans.

vekexasia, 2025-04-20:
You can use the `ventilation_(supply|extract)_only` command. Tbh i don't remember now if it's a timer or it's a permanent set until reset is sent.

Anyways here below a yaml example

```yaml
- platform: template
  switches:
    vmc_supply_ventilation:
      value_template: "{{ states('sensor.comfoair_supply_fan_duty') | int > 0 }}"
      turn_on:
        service: esphome.comfoair_send_command
        data:
          command: ventilation_extract_only_reset
      turn_off:
        service: esphome.comfoair_send_command
        data:
          command: ventilation_extract_only
    
    vmc_extract_ventilation:
      value_template: "{{ states('sensor.comfoair_exhaust_fan_duty') | int > 0 }}"
      turn_on:
        service: esphome.comfoair_send_command
        data:
          command: ventilation_supply_only_reset
      turn_off:
        service: esphome.comfoair_send_command
        data:
          command: ventilation_supply_only
```

Schildkroetenprogrammierer, 2025-04-21:
sorry for misspecifiying my question.
I would like to have both fans set to zero. If there is bad air outside I would like to turn off both fans. If there is a fire inside I would like to turn off both fans.

Currently I can set the state to "absent" but this has a minimum airflow around 50 or 70 m³/h. I would like to have 0 m³/h.