vekexasia/comfoair-esp32 #90: Compile issue with ESPHome 2025.7.3

wauswaus, 2025-07-27:
When compiling with ESPHome 2025.7.3 is get following warnings and compile errors:

Compiling .pioenvs/esp-growatt-modbus/src/main.cpp.o
In file included from .piolibdeps/esp-growatt-modbus/esp32_can/src/esp32_can_builtin.h:40,
                 from .piolibdeps/esp-growatt-modbus/esp32_can/src/esp32_can.h:1,
                 from src/esphome/components/comfoair/comfoair.h:2,
                 from src/esphome.h:25,
                 from src/main.cpp:3:
/data/cache/platformio/packages/framework-arduinoespressif32-libs/esp32/include/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
In file included from .piolibdeps/esp-growatt-modbus/esp32_can/src/esp32_can_builtin.h:42:
/data/cache/platformio/packages/framework-arduinoespressif32-libs/esp32/include/esp_adc/deprecated/include/esp_adc_cal.h:17:2: warning: #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h" [-Wcpp]
   17 | #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h"
      |  ^~~~~~~
Building .pioenvs/esp-growatt-modbus/bootloader.bin
Creating esp32 image...
Successfully created esp32 image.
Generating partitions .pioenvs/esp-growatt-modbus/partitions.bin
Compiling .pioenvs/esp-growatt-modbus/lib4c5/Network/NetworkClient.cpp.o
src/esphome/components/comfoair/comfoair.h: In member function 'virtual void esphome::comfoair::Comfoair::setup()':
src/esphome/components/comfoair/comfoair.h:147:6: error: 'register_service' was not declared in this scope; did you mean 'register_sensor'?
  147 |      register_service(&Comfoair::send_command, "send_command", {"command"});
      |      ^~~~~~~~~~~~~~~~
      |      

When removing the complete Comfoair configuration from the yaml file, all compiles fine.

Tigerstrength, 2025-08-02:
Same issue here. Maybe @vekexasia has an idea about the cause and how to solve this? 

Loving this plugin!

vekexasia, 2025-08-04:
I am looking into this :)

Tigerstrength, 2025-08-04:
Nope it fixed one issue but not all. 

Log is indicating a missing SPI package:
`Library Manager: Installing SPI @ 2.0.0
Warning! Could not find the package with 'SPI @ 2.0.0' requirements for your system 'linux_x86_64' `

Resulting in 
`Compiling .pioenvs/esp-ventilatie/lib805/esp32_can/esp32_can_builtin_lowlevel.cpp.o
Compiling .pioenvs/esp-ventilatie/lib805/esp32_can/mcp2515.cpp.o
.piolibdeps/esp-ventilatie/esp32_can/src/esp32_can_builtin_lowlevel.cpp:32:10: fatal error: esp_intr.h: No such file or directory

******************************************************************
* Looking for esp_intr.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_intr.h"
* Web  > https://registry.platformio.org/search?q=header:esp_intr.h
*
******************************************************************

   32 | #include "esp_intr.h"
      |          ^~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/esp-ventilatie/lib805/esp32_can/esp32_can_builtin_lowlevel.cpp.o] Error 1
.piolibdeps/esp-ventilatie/esp32_can/src/mcp2515.cpp:32:10: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://registry.platformio.org/search?q=header:SPI.h
*
*************************************************************

   32 | #include "SPI.h"
      |          ^~~~~~~
compilation terminated.
*** [.pioenvs/esp-ventilatie/lib805/esp32_can/mcp2515.cpp.o] Error 1
========================= [FAILED] Took 16.69 seconds ========================= `

wauswaus, 2025-08-04:
Tried but unfortunately the same issues

vekexasia, 2025-08-04:
Can you reach out to my discord ? its easier to debug there

vekexasia, 2025-08-04:
Hey @wauswaus , i just tried with n problem with esphome 2027.7.4. 

Are you using Web interface or cli ?

wauswaus, 2025-08-04:
webinterface, Using the home assistant add-on.

vekexasia, 2025-08-04:
i can replicate from webinterface...

dzejPL, 2025-08-12:
I have the same error on ESPHome 2025.7.5

Error logs
```
Library Manager: Installing SPI @ 2.0.0
Warning! Could not find the package with 'SPI @ 2.0.0' requirements for your system 'linux_x86_64'

Compiling .pioenvs/rekuperator/lib4c5/Network/NetworkClient.cpp.o
In file included from .piolibdeps/rekuperator/esp32_can/src/esp32_can_builtin.h:40,
                 from .piolibdeps/rekuperator/esp32_can/src/esp32_can.h:1,
                 from src/esphome/components/comfoair/comfoair.h:2,
                 from src/esphome.h:23,
                 from src/main.cpp:3:
/data/cache/platformio/packages/framework-arduinoespressif32-libs/esp32/include/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
In file included from .piolibdeps/rekuperator/esp32_can/src/esp32_can_builtin.h:42:
/data/cache/platformio/packages/framework-arduinoespressif32-libs/esp32/include/esp_adc/deprecated/include/esp_adc_cal.h:17:2: warning: #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h" [-Wcpp]
   17 | #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h"
      |  ^~~~~~~

Compiling .pioenvs/rekuperator/lib7f9/esp32_can/esp32_can_builtin_lowlevel.cpp.o
.piolibdeps/rekuperator/esp32_can/src/esp32_can_builtin_lowlevel.cpp:32:10: fatal error: esp_intr.h: No such file or directory

******************************************************************
* Looking for esp_intr.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_intr.h"
* Web  > https://registry.platformio.org/search?q=header:esp_intr.h
*
******************************************************************

   32 | #include "esp_intr.h"
      |          ^~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/rekuperator/lib7f9/esp32_can/esp32_can_builtin_lowlevel.cpp.o] Error 1
In file included from .piolibdeps/rekuperator/esp32_can/src/esp32_can_builtin.h:40,
                 from .piolibdeps/rekuperator/esp32_can/src/esp32_can.h:1,
                 from .piolibdeps/rekuperator/esp32_can/src/esp32_can.cpp:1:
/data/cache/platformio/packages/framework-arduinoespressif32-libs/esp32/include/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
In file included from .piolibdeps/rekuperator/esp32_can/src/esp32_can_builtin.h:42:
/data/cache/platformio/packages/framework-arduinoespressif32-libs/esp32/include/esp_adc/deprecated/include/esp_adc_cal.h:17:2: warning: #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h" [-Wcpp]
   17 | #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h"
      |  ^~~~~~~
In file included from .piolibdeps/rekuperator/esp32_can/src/esp32_can_builtin.h:40,
                 from .piolibdeps/rekuperator/esp32_can/src/esp32_can_builtin.cpp:10:
/data/cache/platformio/packages/framework-arduinoespressif32-libs/esp32/include/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
In file included from .piolibdeps/rekuperator/esp32_can/src/esp32_can_builtin.h:42:
/data/cache/platformio/packages/framework-arduinoespressif32-libs/esp32/include/esp_adc/deprecated/include/esp_adc_cal.h:17:2: warning: #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h" [-Wcpp]
   17 | #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h"
      |  ^~~~~~~
========================= [FAILED] Took 62.03 seconds =========================
```

wauswaus, 2025-09-06:
Any updates to mention?

dzejPL, 2025-09-30:
Yeah - works again!
Thanks @vekexasia !

Only some warnings in log - but complilation sucesfull and work :) 

Generating partitions .pioenvs/rekuperator/partitions.bin
In file included from .piolibdeps/rekuperator/esp32_can/src/esp32_can_builtin.h:40,
                 from .piolibdeps/rekuperator/esp32_can/src/esp32_can.h:1,
                 from src/esphome/components/comfoair/comfoair.h:2,
                 from src/esphome.h:26,
                 from src/main.cpp:3:
/data/cache/platformio/packages/framework-arduinoespressif32-libs/esp32/include/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~
In file included from .piolibdeps/rekuperator/esp32_can/src/esp32_can_builtin.h:42:
/data/cache/platformio/packages/framework-arduinoespressif32-libs/esp32/include/esp_adc/deprecated/include/esp_adc_cal.h:17:2: warning: #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h" [-Wcpp]
   17 | #warning "legacy adc calibration driver is deprecated, please migrate to use esp_adc/adc_cali.h and esp_adc/adc_cali_scheme.h"
      |  ^~~~~~~